sharpie7/circuitjs1 #130: Dependent Source Transient Issues

jo3d, 2017-06-21:
When a dependent voltage or current source is present, the simulation of transients does not work at all. The voltage/current in the transient circuit instantaneously changes and the natural response cannot be viewed.

pfalstad, 2017-06-21:
Can you provide an example circuit?  Thanks

jo3d, 2017-06-21:
http://tinyurl.com/yan3xqyw

Here's the circuit I was analyzing. Unless my calculations and theory are incorrect, I don't think the current should instantly change when the switch is closed for a long time then opened.

This proved true for independent sources, but dependent sources seem to give me issues.

Let me know if my calculations are wrong, because then we have a much bigger issue. Thanks for the lightning fast reply!

pfalstad, 2017-06-23:
I'm not really sure what's going on with this circuit, but the VCVS seems to be working fine?  What should be happening?  Could you share your analysis, or make a simpler example that shows what's wrong?

jo3d, 2017-06-23:
When you flip the switch in the circuit I sent you on and off, I don't believe the current shown in the scope should be jumping instantaneously. It should drop over time from 5A to 2A.

sharpie7, 2017-10-21:
This is such a strange circuit that I cannot immediately see how to do a theoretical analysis on the transient, but I can't see anything obviously wrong with the simulation at the in relation to the behaviour of the VCVS

closed 2017-10-21 by sharpie7