sharpie7/circuitjs1 #131: How to have 01 on logic inputs instead of LH

Benur21, 2017-06-24:
Is there a way to set logic inputs to show 0 or 1 instead of L or H without having to go on Circuits>Sequential Logic>Dynamic RAM or another like that and copy them? Also, sometimes when I try to do so, when I paste them their coordinates don't seem to be compatible with the current circuit's ones.

pfalstad, 2017-06-25:
Hi,

Do File->Export as Text, find the lines starting with L (or M for the outputs), and change the 5th number following the L (or M) from a 0 to a 2.  I should make a better user interface for this.

The paste issue might be caused by one circuit having the Small Grid option set, and the other circuit not having it set?

Benur21, 2017-06-25:
Inputs and outputs should have an option to choose between 01 and LH on their Edit menu.
Thank you for telling me about that way to get them.

pfalstad, 2018-01-12:
This is done

closed 2018-01-14 by sharpie7