sharpie7/circuitjs1 #132: When deleting a component using delete, it sometimes delete another random component

Benur21, 2017-06-24:
Sometimes I am hovering, and maybe even clicked a component and when I press Delete it deletes another random component, some which I have probably selected before. This issue can make it complicated to safety remove a component using Delete, and to find there is now some missing component.

pfalstad, 2017-06-25:
Hmm, yeah, if you right-click on a component, then mouse-over another component while the popup menu is showing, then click Delete, it will delete the new component instead of the original.  But all the other menu items (like Cut) don't have this problem.  Weird.

pfalstad, 2017-06-25:
I have a fix for this which I will release soon.

Benur21, 2017-06-25:
Just a question: Is it supposed for Delete to remove the component we are hovering or the last one to be clicked on? I've never understood that.

pfalstad, 2017-06-25:
It's supposed to delete the one you're hovering over..  Or if you've selected multiple items by clicking and dragging, it deletes all of those.

Benur21, 2017-06-25:
But when you select some items, then hover another, all are blue so all look that will be removed.

pfalstad, 2017-06-25:
Yeah, that's not good.  It probably shouldn't highlight the item under the mouse when multiple items are selected.  Also, when the right mouse menu is showing, it should highlight the items that will be acted on (and not any others).

Benur21, 2017-06-26:
Also, right-click on a component, select Edit, then Cancel (or any other to close) then hover/click on another component and Delete. It will delete the first component. However, this doesn't happen when double-clicking a component to open the Edit window.

pfalstad, 2017-06-26:
I can't get that to happen.  But if you right click and copy another component, then right click and try to delete another component, it will delete the first one.  I've got a fix for that too.

Benur21, 2017-06-27:
Gif showing how it happens: [http://html-js.comxa.com/gif-temporary/FalstadDeleteBug.gif](http://html-js.comxa.com/gif-temporary/FalstadDeleteBug.gif)

pfalstad, 2017-06-27:
Oh the delete KEY.  That's a different bug, I'll take a look.

closed 2018-04-24 by Benur21