sharpie7/circuitjs1 #140: Question: how did you derive your bipolar junction transistor model (TransistorElm.java)?

colinhowell, 2017-07-09:
This is a question for Paul Falstad, since the code is his and dates back to the original Java simulator.

I've been looking over the transistor model in TransistorElm.java, and I have to admit I'm confused. The model mostly looks like the standard Ebers-Moll model, but there appear to be some changes that I don't understand. (I'm not counting the change in the sign convention for the emitter current or the adjustments done in the event of convergence failure.) In particular, what are the intended roles of "leakage", "fgain", and "rgain"? I've been assuming they are meant to correspond to the Ebers-Moll model's parameters I_S ("saturation current"), alpha_F ("forward common-base current gain", alpha_F = beta_F / (beta_F + 1)), and alpha_R ("reverse common-base current gain", alpha_R = beta_R / (beta_R + 1)). But they aren't used in the same way as those parameters in that model. Also I'm wondering why for vbe < 0 the currents are clamped to their vbe = 0 values.

I'm not ready to file a bug yet, since I'm not sure what your intentions were, and I haven't yet devised a scenario which clearly shows a problem. I just felt I needed to sound you out first.

pfalstad, 2017-07-09:
Hi,

yep, leakage, fgain, and rgain are I_S, alpha_F, and alpha_R.  The model I used is taken from this book: https://www.amazon.com/Electronic-Circuit-System-Simulation-Methods/dp/0070501696

I'm not why the currents are clamped for vbe < 0.  Should be worth a comment at least!  I had a lot of trouble getting this to work and maybe that is one of my attempts at fixing a convergence issue.  There is similar code in MosfetElm.java to avoid numerical issues (which is properly commented).

colinhowell, 2017-07-09:
Thanks for the quick reply!

Unfortunately I don't have easy access to Pillage, so I can't check things there myself, though I do get the impression that it's a highly regarded reference. (Supposedly the Stanford libraries have it, but upon further checking I found it was in off-campus storage. :( )

Anyway, I'm still thinking about how I can confirm and demonstrate where I think the error may lie.

colinhowell, 2017-07-12:
Sorry to take so long to get back to you on this. I was sure there was a problem, but first had trouble verifying it, and then was under the incorrect impression that the problem was merely one of proper scale, making it harder for me to argue why a change was justified. But I've finally found an (admittedly somewhat contrived) case where the discrepancy between your model and the standard Ebers-Moll one causes **qualitatively** different behavior, independent of scale.

Here I don't mean the clamping of currents for vbe < 0 or the convergence-fixing adjustments. I haven't even considered those yet, and the case I crafted deliberately avoids them. The problem is in the basic current equations in TransistorElm's `doStep()` method. Here are your equations:

```
	    ie = pnp*leakage*(-(expbe-1)+rgain*(expbc-1));
	    ic = pnp*leakage*(fgain*(expbe-1)-(expbc-1));
	    ib = -(ie+ic);
```

From my reading of the Ebers-Moll model as described by various references (accounting for your different convention for the positive direction of `ie`), and your confirmation that `leakage`, `fgain`, and `rgain` correspond to that model's I_S, alpha_F, and alpha_R, I've concluded that the equations should actually read:

```
	    ie = pnp*leakage*(-(1/fgain)*(expbe-1)+(expbc-1));
	    ic = pnp*leakage*((expbe-1)-(1/rgain)*(expbc-1));
	    ib = -(ie+ic);
```

The specific changes are:

- for `ie`, the `(expbe-1)` term acquires a factor of `(1/fgain)` and the `(expbc-1)` term loses a factor of `rgain`;
- for `ic`, the `(expbe-1)` term loses a factor of `fgain` and the `(expbc-1)` term acquires a factor of `1/rgain`.

This makes the equations consistent with the standard Ebers-Moll equations.

Since I haven't seen the relevant section of Pillage, I'm not sure how you got the forms you've actually used. I have a suspicion, but it might be wrong. If you treat the saturation currents of the base-emitter and base-collector junction diodes in the Ebers-Moll model as **separate** quantities, with parameters I_SE and I_SC, the resulting equations do look like yours. (This can be seen in several references on the Ebers-Moll model; for example, see the very top of http://people.seas.harvard.edu/~jones/es154/lectures/lecture_3/bjt_models/ebers_moll/ebers_moll.html; alternatively, see https://ecee.colorado.edu/~bart/book/book/chapter5/ch5_3.htm#5_3_3, figure 5.3.3 and equations 5.3.14 and 5.3.16, which says the same thing using different notation for I_SE and I_SC.)

If you used `leakage` for **both** I_SE and I_SC, you'd get the present equations. The problem is **you can't do that**. The diode saturation currents I_SE and I_SC are not independent of the gains alpha_F and alpha_R. They are related by another equation called the "reciprocity relation": I_SE * alpha_F = I_CS * alpha_R. (Again, this is mentioned in many references on the Ebers-Moll model; for example, see the reference to S.M. Sze on the Harvard page, or see equation 5.3.17 on the University of Colorado page.) Both sides of the reciprocity relation are commonly renamed as the single parameter I_S (your `leakage`; SPICE calls this the "transport saturation current"), and that gives the form of the Ebers-Moll equations I'm suggesting.

The change in the equations leads to quite different behavior for a saturated BJT if V_CE goes to 0. Under those conditions, both the base-emitter and base-collector junctions will have the same forward bias, so both junctions conduct in that direction. The incoming base current will rise to much higher levels, exiting through both the collector and the emitter. In your simulator, if `beta` is large most of this current leaves through the **emitter**. With the equations changed as I am suggesting, as V_CE goes to 0, for large `beta` most of the incoming base current leaves through the **collector**. (Also the limiting value of the base current is twice that of the unmodified simulator, all other things being equal.)

This configuration can be simulated in SPICE using its default NPN model, which reduces to Ebers-Moll as long as most of the default parameters aren't changed. The result matches that I see in circuitjs1 with my suggested equation changes.

Here is an example circuit which demonstrates the behavior I'm talking about. It has a single NPN transistor with its base held at +0.35 V and its emitter held at -0.35 V. (With V_BE fixed at +0.7 V, `DoStep()` will never clamp `expbe`.) The collector is tied to an AC voltage varying slowly between +0.35 V and -0.35 V with a 40 Hz frequency.

With the collector at +0.35 V, V_BC is zero, V_CE is +0.7 V, and the transistor is forward active (though only technically, depending on how that mode is defined). The current entering the collector is `beta` times the base current, and the current leaving the emitter is the sum of the collector and base currents.

When the collector voltage falls, V_BC > 0 and the transistor becomes saturated. As the collector goes to -0.35 V, V_BC goes to +0.7 V, the same as V_BE, and V_CE goes to zero. The base current grows dramatically. The question is, where does it go?

The Ebers-Moll model says the emitter current plummets, and the collector current **reverses**, becoming strongly negative. Most of the base current is exiting through the collector. SPICE agrees. So does the simulator with my fixed version of the equations for current in a BJT. But with the original equations, although the collector current still reverses, its magnitude falls to a far lower  [... cut ...] PICE default temperature of 27 C, and SPICE's NPN model's IS parameter has to be made equal to circuitjs1's `leakage` (However, the currents are directly proportional to IS, so one can simply mentally correct for the ratio of IS to `leakage`.)

I'm not ready to suggest this equation problem should be corrected right away, because I'm still wondering whether it might affect your convergence problems or your fixes for them. Are there any circuits that were particularly troublesome for convergence?

pfalstad, 2017-07-12:
Very thorough.  Thanks!!  Pillage does indeed treat I_SE and I_SC as separate quantities, but it failed to mention the relation between them, so I naively assumed the same value would work for both.

This circuit is one I remember as being particular problematic for convergence: http://www.falstad.com/circuit/e-multivib-bi.html .  But if we change the transistor model, and it sounds like we should, we'll have to test everything.

colinhowell, 2017-07-13:
Hmm, I'm beginning to wonder whether in my focus on the `ie` and `ic` equations, I may have neglected something. If the equations for `ie` and `ic` as functions of `vbe` and `vbc` are changed, shouldn't that change the equations for `gee`, `gec`, `gce`, and `gcc` as well? Those are conductances, right? I don't have the theory in front of me and haven't worked things out fully, but at first glance they look like `d(ie)/d(ve)`, `d(ie)/d(vc)`, `d(ic)/d(ve)`, and `d(ic)/d(vc)`.

pfalstad, 2017-07-13:
Yes, they're conductances.  gec is d(ie)/d(vbc), gcc = d(ic)/d(vbc), etc.

colinhowell, 2017-07-13:
Wow, a quick-and-dirty test of correcting those conductances (which actually only seemed to require changing a couple of factors in the `gee` and `gec` equations) actually made the bistable multivibrator's convergence a hell of a lot worse. Well, you did say it was a difficult case, and I just wanted to try it out of curiosity. Anyway, I had already figured that I would have to think very carefully about what's happening in the math to be at all useful on the convergence problem.

pfalstad, 2017-07-14:
have you tried using the same equations, but two different leakage currents (leakageSE and leakageSC)?  less math that way..

pfalstad, 2017-07-14:
what did you get for the conductances?

pfalstad, 2017-07-14:
I got

gee = -leakage*vdcoef*expbe/fgain
gec = leakage*vdcoef*expbc
gce = leakage*vdcoef*expbe   (or -gee*fgain, same as before)
gcc = -leakage*vdcoef*expbc/rgain  (or -gec*(1/rgain), same as before)

colinhowell, 2017-07-14:
Yeah, that's exactly what I got.

I suspect some other issue, like a previous fix for the convergence, is now causing problems. But that's only a suspicion at the moment.

Unfortunately I haven't been able to spend much time thinking about this the last few days. Outside life is demanding my attention right now.

colinhowell, 2017-07-14:
For the moment, though, I think it's better to let things stay as they are until the consequences of correcting the transistor equations and the causes of the convergence difficulties are better understood. As you said, you had a lot of trouble getting this working. Right now it **does** work, to the extent that none of the existing circuits show any obvious anomalies, and the code seems to be fairly brittle, so a cautious approach seems wise.

colinhowell, 2017-07-25:
OK, coming back to this problem. I checked the bistable multivibrator circuit with the corrected model again, with correct conductances as we'd discussed earlier. While earlier I had reported that the convergence was "a hell of a lot worse", I believe I was actually misled by an unrelated effect. With the circuit in its as-loaded configuration, each state flip (from reset to set or vice versa) requires 42 subiterations for convergence, versus 22 iterations for the original incorrect model. This convergence is indeed slower, but that's not what I was reacting to before.

Sometimes this circuit will get wedged in a both-outputs-low state (actually, both outputs are at an intermediate voltage, but closer to the low end). Indeed, the circuit will definitely end up in this state if it is reset, rather than reloaded. **This state can occur regardless of whether the corrected or uncorrected transistor model is used.** When in this state, each push of a state-change switch requires 250 subiterations to converge, and the circuit will probably stay wedged. In fact, the uncorrected model appears rather more likely to spontaneously end up in the wedged state than the corrected model, and the latter seems more likely to "unwedge" the circuit even when it is wedged, though I'm not certain whether this is a reliable difference or just my good luck in my testing.

sharpie7, 2017-07-25:
My experience is that these bistable circuits can be tricky to simulate without the natural bias to one-side which would occur due to normal tolerances of components. Tjey can get wedged in states which are the equivalent of a cone balancing on its point. If you unbalance the circuit slightly (say changing one of the collector resistors to 101Ohms) does that help address the sticky state?

colinhowell, 2017-07-25:
From a little testing, doesn't seem to help that much. I tried unbalancing different components, including changing one transistor's gain, since I understand those are generally poorly controlled. Sometimes there was no obvious effect, sometimes it helped balance in one direction but not the other (say "from clear to set" and not "from set to clear"). A full circuit reset seems to always come up in the wedged midpoint zone I mentioned.

Actually I'm wondering if the analogy here is not so much balancing a cone on its point as setting a tall glass on a table. The circuit, as constructed, may find that middle state quite stable and require quite a strong push to send it one way or the other. It only comes up in a well-defined "clear" state because it loads in an unbalanced state to begin with. That might explain why small changes in component values have little effect on this.

pfalstad, 2017-07-25:
There already is a natural bias to one side..  The resistor on the top right is 320 and the top left one is 330.  I seem to remember that the circuit wouldn't work at all without that; but when I tried it now, it works either way, for me.

So you're saying this state is even more likely to happen with the uncorrected model?  How long do you have to run it?  Because I'm not seeing it happen.

colinhowell, 2017-07-25:
Paul, from the component values you just gave, I think you're looking at the astable multivibrator, not the bistable one. Unless your bistable one is different from the one on your website?

pfalstad, 2017-07-25:
Oops, I sent you the wrong circuit.  The astable multivibrator is the one that I had convergence issues with.

Yes, I see that if you reset the bistable multivibrator, then it gets in a weird state.  I've never tried that before.  I'm guessing that bug's been there forever.

pfalstad, 2017-07-25:
Actually in the java version, if you reset the bistable, it gets stuck in a both-outputs-low state but you can easily get out by clicking either of the logic inputs to set or reset it.  That doesn't work in the javascript version.  It did work in 2.1.3js but not in 2.1.4js, so something got broken around there; probably caused by "be more aggressive in modifying transistor model to help convergence".

colinhowell, 2017-07-26:
Yeah, I just checked it; that change was indeed responsible.

(For the following, I'm not using the corrected model; I just wanted to see the impact of the aggressive convergence change.)

The version before the change took longer to converge from the both-outputs-low state after a logic-input click: 1421 subiterations. But it did converge, and the added parallel conductances *gmin*, though significant, were relatively low, equivalent to resistors of around 55 kilo-ohms.

After the change, convergence from both-outputs-low with an input pulse is a lot faster: 250 subiterations. But *gmin* increases so fast that there's no hope of exiting the both-outputs-low state before convergence is reached: its final value at convergence is equivalent to resistors of around 32 **ohms**. The transistors leak so badly that they cease to be effective.

pfalstad, 2017-07-26:
you could change the 300 to a 600, that seems to work.

colinhowell, 2017-07-26:
Just for the sake of it, I tested the astable vibrator with the current model (uncorrected, with aggressive convergence). It too may need a large number of subiterations to converge (265) when one transistor switches off and the other switches on. (This first happens one full cycle, or two half-waves, after startup.) Convergence after 265 subiterations means that, for that one timestep, both transistors have an added *gmin* equivalent to 11-ohm resistors across each P-N junction.

colinhowell, 2017-07-28:
So, I've been testing out the corrected model (with both the equations for current and those for the conductances changed), together with some other tweaks to the model, and I have a proposal (though not yet a pull request; I'm not yet ready for that).

First, I eliminated:

    if (expbe < 1)
        expbe = 1;

which arbitrarily clamps the effect of the base-emitter voltage as if that junction can never be reverse-biased. My guess is that this was to avoid extremely low conductances and singular results, but I have an idea to reach the same end by different means.

Instead of waiting until 100 subiterations before increasing _gmin_ from zero, I'm starting with it at a small **nonzero** value, say 1.0e-12 or 1.0e-13. (SPICE2 and SPICE3 also have a GMIN parameter, which is likewise a minimum conductance placed parallel to all PN junctions; its default value is 1.0e-12.) Then immediately start ramping it up, as you do now, but starting from the smaller initial value. You might use a denominator in the exponential of 300, or 600, or 3000, depending on what you want the worst case to be. I've found that with 300, I get convergence for sharp edge transitions in at most 280 subiterations with the most problematic circuits (the astable multivibrator, monostable multivibrator, transistor Schmitt trigger, joule thief, and tests/square67.txt). With 600, convergence takes longer in the edge cases, sometimes over 500 subiterations. In both cases, you can't guarantee that the parallel conductance won't become unrealistically large, but from what I can tell, that seems to be an unavoidable drawback of how you determine convergence here.

SPICE3 does a form of conductance stepping for convergence as well. Although this is also called "GMIN stepping" in the SPICE manual, according to the SPICE3 paper it is actually independent of the small conductance GMIN which is always added parallel to PN junctions. SPICE3's stepped conductance is placed between every node and ground, and it is stepped in the opposite direction: when it activates (after at least 100 iterations), it starts at a **large** value to speed up convergence, and then **decreases** as the solution converges, eventually being reduced to zero for the end solution. The idea is that the large conductance helps the Newton-Raphson solver settle in the correct neighborhood more quickly, before the conductance is then tightened so the solution converges on the correct value. This sounds like it would be more accurate, but of course making it work would require a major transformation of the simulator's logic, probably more than you'd want to do at this point.

colinhowell, 2017-08-07:
Sorry it's been so long since I've said anything. I've actually gotten a bit off track, trying to better understand why the most problematic circuits (e.g. astable multivibrator, Schmitt trigger, etc.) have such trouble converging.

But right now I'm a bit puzzled by something else: how the stamps in TransistorElm's doStep() method handle _gmin_. I understand that the basic formulas for those stamps came from Pillage, but that seems to be before you added _gmin_. I don't quite understand the _gmin_ values in the matrix: they don't seem to match how I'd think two conductances parallel to the P-N junctions should appear. Perhaps my thinking is wrong, but I also notice _gmin_ isn't added on the right hand side at all.

When I deliberately hacked the code to completely disable the transistor current formulas, so that a TransistorElm's **only** contribution to the circuit was through its _gmin_, and set _gmin_ to a relatively large fixed value (0.1, or 10 ohms), and then examined the simple NPN transistor circuit, I saw peculiar, inconsistent results. The currents reported by the transistor itself were 0, as I expected, because it just reports the ic and ib values, and ic and ie were both set to 0, so naturally ib would be 0 as well. The currents leaving or entering the voltage sources and ground, as reported by those elements, had the expected values and direction. (As comparison, I used a similar circuit with the "transistor" replaced by two resistors of the proper value and arrangement.) But the wires attaching those elements to the "transistor" also reported 0 current. Very odd. 

Furthermore, if I replaced the wires with elements which are equivalent to wires, like ammeters, 0-volt voltage sources, labeled connections, or very small resistors, these all report the expected amount of current. I assume this is because all of these, except for very small resistors, are implemented as 0-volt voltage sources, rather than being optimized like wires are. Only wires show anomalous behavior.

Going further, if I supply both the "transistor" circuit's "collector" and the corresponding in the dual-resistor comparison circuit from the same voltage source, using wires for the connection, and drain off current to a common ground from both "emitters", again using wires, the common collector wire shows the combined current going to the resistor circuit, but the currents otherwise look shared equally, and the common emitter wire shows only the resistor circuit's share going to ground.

All very weird.

I'm not exactly sure what the cause of this inconsistency is, but I wonder if it could be interfering with convergence somehow?

pfalstad, 2017-08-07:
I do think gmin should be added to the right side.  It should probably be added into gee/gec/gce/gcc as appropriate if possible.  It sounds like that is causing some of the strange behavior, and probably could cause problems with convergence too.

The problem with the wires reporting zero current is caused by a wire removal optimization the simulator does.  The wires are not part of the simulation at all; they are optimized out.  So they get their current from adjacent elements whenever we need to display them.  If the elements (like the transistor) have the wrong current, so will the wires.  I don't think this issue will affect convergence.

I don't totally understand the "going further" paragraph, but it sounds like the same issue with the wires.

colinhowell, 2017-08-08:
Yeah, I understand what's happening now. Any current passing through the _gmin_ conductances is not added into the current variables _ic_, _ie_, and _ib_, and it is those variables which are used to find the currents of adjoining wires. This doesn't affect other wire-like elements which don't have the wire optimization. All the weird effects I just reported are due solely to that.

(The "going further" paragraph would, frankly, have been clearer if I'd bothered to show the test circuit I made, and it didn't help that I left a word out.  But as you said, it is indeed the same issue.)

Actually, I apologize; my understanding of how _gmin_ should affect the stamps was wrong. I just worked the math out and got exactly the same expressions you did. In particular, the _gmin_ terms on the right hand side would be equal and of opposite signs, so they cancel each other out. That's why _gmin_ doesn't appear there.

I no longer think this affects convergence at all. The only effect is that the currents that are displayed by the transistor's terminals (or wires attached to them) can differ from those shown by adjoining non-wire elements. This becomes most obvious when watching transistor circuits which need heavy help from _gmin_ to ensure convergence in reasonable time. The timesteps with the greatest number of subiterations look normal if you examine the transistor's view of its currents, but if you look at adjacent non-wires, they show up as current spikes, corresponding to very high _gmin_ values.

I suppose that the reason for not including _gmin_ explicitly when calculating the currents was to hide this effect?

colinhowell, 2017-08-08:
OK, I've just bitten the bullet and filed pull request #149 for the TransistorElm model fixes. I have tested these on all the example circuits which use that component, and haven't had any problems. The patch is pretty straightforward, since I avoided changing more than I felt necessary.

colinhowell, 2017-08-11:
Since the PR has been accepted, should we close this issue now?

sharpie7, 2017-08-14:
Thanks! Closed

closed 2017-08-14 by sharpie7