sharpie7/circuitjs1 #144: Feature: two coil latching relay

LittleGitBeginner, 2017-07-20:
Is it possible to create a two coil latching relay?

Best regards.

li-zhenhai, 2023-10-17:
[image]
`/*    
    Copyright (C) Paul Falstad and Iain Sharp
    
    This file is part of CircuitJS1.

    CircuitJS1 is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 2 of the License, or
    (at your option) any later version.

    CircuitJS1 is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with CircuitJS1.  If not, see <http://www.gnu.org/licenses/>.
*/

package com.lushprojects.circuitjs1.client;

import com.google.gwt.user.client.ui.Button;
import com.lushprojects.circuitjs1.client.util.Locale;

// 0 = switch
// 1 = switch end 1
// 2 = switch end 2
// ...
// 3n   = coil
// 3n+1 = coil
// 3n+2 = end of coil resistor

class RelayElm extends CircuitElm {
    final int FLAG_SWAP_COIL = 1;
    final int FLAG_SHOW_BOX = 2;
    final int FLAG_BOTH_SIDES_COIL = 4;
    int[] voltSources = new int[2];
    double inductance;
    Inductor ind;
    Inductor ind2;
    double r_on, r_off, onCurrent, offCurrent;
    Point coilPosts[], coilLeads[], swposts[][], swpoles[][], ptSwitch[];
    Point lines[];
    Point outline[] = newPointArray(4);
    double coilCurrent, switchCurrent[], coilCurCount, switchCurCount[];
    double coilCurrent2;
    // fractional position, between 0 and 1 inclusive
    double d_position;

    // integer position, can be 0 (off), 1 (on), 2 (in between)
    int i_position;

    double coilR;

    // time to switch in seconds, or 0 for old model where switching time was not constant
    double switchingTime;

    int poleCount;
    int openhs;
    boolean onState;
    final int nSwitch0 = 0;
    final int nSwitch1 = 1;
    final int nSwitch2 = 2;
    int nCoil1, nCoil2, nCoil3;
    int nCoil4, nCoil5, nCoil6;
    double currentOffset1, currentOffset2;

    public RelayElm(int xx, int yy) {
        super(xx, yy);
[... 499 lines cut ...]
        if (n == 9 && ei.value > 0)
            switchingTime = ei.value;
    }

    boolean comparePairNew(int a1, int a2, int b1, int b2) {
        return ((a1 == b1 && a2 == b2) || (a1 == b2 && a2 == b1));
    }

    boolean getConnection(int n1, int n2) {
        if (comparePairNew(n1, n2, 3 * poleCount, 3 * poleCount + 1))
            return true;
        if (comparePairNew(n1, n2, 3 * poleCount + 2, 3 * poleCount + 3))
            return true;
        if (n1 >= 3 * poleCount || n2 >= 3 * poleCount) {
            return false;
        }
        return (n1 / 3 == n2 / 3);
    }

    int getShortcut() {
        return 'R';
    }
}
    
`

LittleGitBeginner, 2023-11-04:
Thx.
With 2 antiparallel virtual diodes also can simpley sumulated an one coil (polarisied) latching relay.