sharpie7/circuitjs1 #153: Feature - DC-Power as alternative to RMS Power

JJeppe, 2017-08-16:
http://tinyurl.com/yb8tnw35 
A Circuit with a 100W lamp - powered with either 230V DC or 230V AC RMS

This example will hopefully explain my issue.
When the lamp powered by a DC source will the energy consumed (finally) be 100W
However when the lamp powered with the same equvivalent AC 230V RMS will the (MAX) Power consumed be 200W
If I choose a RMS value will the Power consumed the result be 132.36 W.
I would like to have a "DC-average" value as well, giving me the result 100W for a 230V RMS AC

pfalstad, 2017-08-16:
I don't understand what you're asking here.  Do you want something changed in the lamp, or the AC source, or the DC source, or the scope?

JJeppe, 2017-08-16:
Hi Paul
Thanks for a great simulator which make it easier to explain students about Electronics.

My hope was that the last line explained my "problem".
The idea behind RMS is the find an equvivalant DC value - hence 230V DC = 230V AC RMS => Power consumed in lamp = 100W.
The Power scope is fine with DC values - however will the display show MAX Power = 200W with AC values - I would like the have "DC-Average value" (http://www.electronics-tutorials.ws/accircuits/average-voltage.html) 
This should give  a reading = 100W instead of the 200W (and Show RMS average not useful here)
Please note! For a pure AC value will the DC-Average value be 0 (Zero)

pfalstad, 2017-08-16:
Right, so you want the scope to show the average power, not just the max power.  That's a good idea.

JJeppe, 2017-08-16:
Thanks - looking forward

endolith, 2018-01-08:
Currently the scope has an option for "Show RMS Average" even when "Show Power Consumed" is active, and will show "mWrms", which is a nonsensical measurement:

[image]

It should be "Show Average" when "Show Power Consumed" is active, and the units should just be "mW" or "mW(avg)".

https://github.com/sharpie7/circuitjs1/issues/115 is a duplicate, I think.

sharpie7, 2018-01-27:
Fixed in latest version.

closed 2018-01-27 by sharpie7