sharpie7/circuitjs1 #157: mosfet's Rds is too high

yunuzk, 2017-08-27:
Hi, i tried to simulate mosfet transistor, i supplied the gate about 5-10 volt,
the D-S resistance is about several ohm.
I calculate the Rds = Vds / Ids.
The actual Rds of actual mosfet generally about 10-100 mili ohm (0.010-0.100 ohm).
for low powered application that's ok, but for several watt it is inaccurate.

there are no option for Ids, only threshold voltage.

pfalstad, 2017-08-27:
hi,

you can adjust the "beta" parameter to change Rds.  set beta = 1/(RdsON*(Vgs-Vt))

OmegaHaxors, 2017-09-18:
Most modern MOSFETs have a Beta higher than 2 if my calculations are correct. Considering the default value in the sim is 20m, it just goes to show how far we've come in MOSFET technology since this program was originally made.

emilfihlman, 2018-08-22:
If most modern mosfets have a beta of around 800m, then something is still broken with the model. A very simple p80nf55 should conduct many tens of amperes at 6Vds and 6Vgs, yet at 800m it conducts barely 10A.

pfalstad, 2018-08-22:
A beta of around 35 is more appropriate for that device.

pfalstad, 2018-08-22:
Ordinary MOSFETs like the ZVP3306A have a beta of around .027.  Power MOSFETs have beta that are much higher.

closed 2020-09-10 by pfalstad