sharpie7/circuitjs1 #166: Inconsistent Behavior with MOSFETs

OmegaHaxors, 2017-09-17:
http://tinyurl.com/y8usxn8e

pfalstad, 2017-09-18:
Hi, could you expand on this.  What do you see happening, and what do you think should be happening?

OmegaHaxors, 2017-09-18:
When circuit 1A is grounded, it drives a current. 1B shows that no current is driven when not grounded. 2A shows that nothing happens if the bridge is reversed and 2B confirms this. 

This is inconsistent because despite being built the same way with minor changes, only the grounded 1A circuit exhibits this behavior, leading me to believe that it's an error in the calculation and not a factor of the FET itself.

pfalstad, 2017-09-18:
In circuit 1A, since the source voltage > drain, drain and source are swapped.  The mosfets in this simulator are symmetric since there's no body terminal or body diode.  I believe someone has filed a separate issue about that.

Since the gate is higher than the drain (acting as the source) by more than 1.5 V, the mosfet conducts.  Because of where you put the ground node, the drain is at -2.5V.  The battery is split like you said.  The gate is not connected to anything that would define its voltage so it is grounded by default.

In the case of 1B and 2B, since there's no ground connection, the simulator picks a more-or-less random node to tie to ground.  In this case, the gate is not higher than drain or source by the threshold voltage, so it doesn't conduct.

OmegaHaxors, 2017-09-19:
It appears this is just a fundamental property of how MOSFETs work, and is not a simulator artifact; as inconsistent as it might seem, it does make logical sense. This is going to lead to some pretty interesting designs now that I know about it, and i'm going to try my best to get around the limitation. Thanks for your time.

OmegaHaxors, 2017-09-19:
Wait a minute... Wouldn't swapping the Drain and Source have a meaningful impact in this instance? Since moving the Bulk away from the gate would prevent it from having an influence on it and thus blocking the current?

closed 2017-09-19 by OmegaHaxors