sharpie7/circuitjs1 #167: Request feature: thermal view

yunuzk, 2017-09-17:
It is better if the simulator have an thermal view, so the temperature/power leakage can be observed.

OmegaHaxors, 2017-09-18:
Go over to options and click Show Power. It's not perfect but it will give you a good approximation on how much power is being bled away by your resistors. Green means power is being created (like from a capacitor or source) and Red means it's being destroyed. Slide the Power Brightness for more clear results.

sharpie7, 2017-09-18:
@OmegaHaxors If you hover the mouse over a component then the info box on the bottom right of the main panel will (normally) show the instantaneous power dissipation ("P=....")

OmegaHaxors, 2017-09-18:
It won't on the various transistors, Voltage Sources, and the digital switches. You don't realize how much you appreciate its existence until it's not included on the part you want to examine.

sharpie7, 2017-10-21:
Power info added for transistors, diodes and DC sources in latest version on my site.

closed 2017-10-21 by sharpie7