sharpie7/circuitjs1 #169: #Feature

scheng123, 2017-09-19:
#Feature: Can you stablize the pwn signal when adjusting duty cycle? Something like this: 
[image]

scheng123, 2017-09-19:
Sorry, github didn't let me upload gifs, here's the original source: https://www.youtube.com/watch?v=YmPziPfaByw ; First 4 seconds of the video.

scheng123, 2017-09-19:
Can you not make the signal move to the left? It always confuses me when I am trying to compare the duty cycle.

sharpie7, 2017-09-19:
@scheng123 for the last point you can just hit the "RUN/Stop" button to freeze the scopes if you want to take measurements.

I have updated the title to reflect the PWM point.

scheng123, 2017-09-19:
Yes, but I don't want to keep stopping and starting, and using a ruler, for every cycle, can you just "lively" take measurements for the duty cycle? Without the scopes moving? And doesn't hitting the "Run/Stop" stop the scope entirely, I don't want to continuously hit the Run/Stop button!