sharpie7/circuitjs1 #170: Request feature: schottky diode

ThePicMan, 2017-09-27:
Hi,
I thought that it would be sufficient to edit Vforward, but I get completely wrong results.

Would it thus be possible to add a schottky diode simulation?

Thank you very much.

TPM

pfalstad, 2017-09-30:
Hi, I would have thought editing Vforward would be enough.  What specific device are you trying to emulate?  What Vforward did you try, and what was wrong with the results?

Iacob-Cezar-Catalin, 2017-10-18:
the problem with diodes is that there are many kinds of diodes and each kind has it's own U vs I curve, it's own parasitic capacitance that changes with the voltage applied when the diode is reverse-biased (can't simulate that with a capacitor because there is no voltage controled capacity component), each diode has it's own heat curve... there are many variables...

Iacob-Cezar-Catalin, 2017-10-18:
Like, for example, zenner diodes have a reverse bias "knee" that is at 90* downwards if the zenner is ideal but, there are real zenners out there with a more smooth curve. There are varactor (varicap) diodes that are used because they have an enhanced variable parasitic capacitance. Schottky diodes have a really low parasitic capacitance but they have a high reverse leakage current that goes up a lot with temperature (up to a few miliamps for big diodes). Tunnel diodes also should be more diverse...

ThePicMan, 2017-12-27:
Hello,
I wrote 2 eMails but they aren't appearing here. In the past if I replied via eMail, things showed up in the site as well, I don't know why it didn't work now, but as I noticed it I will re-post here the messages I had eMailed:

---

Hello Paul,

for example, try this simple circuit:

$ 1 0.000001 28.90693621245522 79 5 43
r 752 416 688 416 0 15000
x 807 650 917 653 4 24 Neg\spulse
x 809 261 916 264 4 24 Pos\spulse
g 960 592 960 656 0
r 688 480 752 480 0 15000
w 752 416 752 432 0
w 752 480 752 464 0
p 752 432 752 464 1 0
w 688 416 688 432 0
w 688 480 688 464 0
p 688 432 688 464 1 0
r 816 416 816 480 0 30000
w 752 416 816 416 0
w 816 416 880 416 0
w 752 480 816 480 0
w 816 480 880 480 0
w 928 416 928 480 0
w 880 480 928 480 0
w 960 480 960 416 0
w 928 592 960 592 0
w 928 304 960 304 0
w 960 592 1008 592 0
z 1008 592 1008 304 1 0.805904783 3.3
w 880 416 880 272 0
w 880 592 880 624 0
w 960 416 880 416 0
w 960 304 1008 304 0
w 688 480 592 480 0
w 688 416 592 416 0
v 592 480 592 416 0 1 40 20 0 0 0.5
w 880 480 880 592 0
d 928 416 928 304 1 0.805904783
d 928 592 928 480 1 0.805904783
d 960 592 960 480 1 0.805904783
d 960 416 960 304 1 0.805904783
d 272 464 288 464 1 0.805904783
o 10 64 0 4098 40 0.1 0 1
o 23 64 0 12290 3.6353368554261936 0.0001 0 2 23 3
o 24 64 0 12290 3.6353368554261927 0.0001 0 2 24 3

it clearly works. Now, if you edit the Vforward of the 4 diodes in the bridge from 0.805904783 Volt to e.g. 0.4, it will not work anymore as expected, it's like if reverse current is nearly as high as forward current now, and the bridge is nearly shorted (see voltages):

$ 1 0.000001 28.90693621245522 79 5 43
r 752 416 688 416 0 15000
x 807 650 917 653 4 24 Neg\spulse
x 809 261 916 264 4 24 Pos\spulse
g 960 592 960 656 0
r 688 480 752 480 0 15000
w 752 416 752 432 0
w 752 480 752 464 0
p 752 432 752 464 1 0
w 688 416 688 432 0
w 688 480 688 464 0
p 688 432 688 464 1 0
r 816 416 816 480 0 30000
w 752 416 816 416 0
w 816 416 880 416 0
w 752 480 816 480 0
w 816 480 880 480 0
w 928 416 928 480 0
w 880 480 928 480 0
w 960 480 960 416 0
w 928 592 960 592 0
w 928 304 960 304 0
w 960 592 1008 592 0
z 1008 592 1008 304 1 0.805904783 3.3
w 880 416 880 272 0
w 880 592 880 624 0
w 960 416 880 416 0
w 960 304 1008 304 0
w 688 480 592 480 0
w 688 416 592 416 0
v 592 480 592 416 0 1 40 20 0 0 0.5
w 880 480 880 592 0
d 928 416 928 304 1 0.4
d 928 592 928 480 1 0.4
d 960 592 960 480 1 0.4
d 960 416 960 304 1 0.4
d 272 464 288 464 1 0.805904783
o 10 64 0 4098 40 0.1 0 1
o 23 64 0 12290 0.07218356945567724 0.0001 0 2 23 3
o 24 64 0 12290 0.07218356945567724 0.0001 0 2 24 3

Cheers,
TPM

---

Hello Iacob,

Perhaps because it's web-based, perhaps because it's so nice, but I find myself brainstorming more and more with Paul Falstad's applet than with regular Spice software. So the addition of Schottky diodes and, why not, Varicap, Tunnel, etc.. later, would certainly be welcome. I.e. do not make this simple and nice software more complex, but more articulated, yes, please.

---

ThePicMan, 2018-02-23:
Hello, no news about this thread?

BrandonLWhite, 2018-10-24:
There's definitely something off with diodes as you lower the forward voltage. They begin to pass current in the reverse direction.  For instance, lowering from 0.6 to 0.25 and the diode passes significant current in the reverse direction when the diode is reversed biased.

This makes simulating simple Schottky diode voltage limiters not possible.

BrandonLWhite, 2018-10-24:
This demonstrates the issue concisely.  Notice how much current is flowing in the reverse direction in the diode.  I merely changed Vf to 0.25 V.  If you raise it to the default 0.6 V it will block the current.
```
$ 13 0.000005 10.20027730826997 50 5 50
R 304 304 224 304 0 0 40 24 0 0 0.5
r 384 304 544 304 0 3900
R 544 160 544 80 0 0 40 3.3 0 0 0.5
w 608 304 544 304 0
p 608 304 688 304 1 0
r 544 304 544 432 0 1000
g 544 432 544 464 0
d 544 304 544 160 1 0.25
s 304 304 384 304 0 1 false
w 688 432 544 432 0
o 4 64 0 4098 5 0.1 0 1
o 1 64 0 4099 0.0000762939453125 0.00009765625 1 2 1 3
38 0 0 3.3 24 Voltage
```

pfalstad, 2018-10-25:
Wow that's pretty bad.  It seems diodes aren't modeled correctly when the forward drop is changed..  more parameters may be needed to do it properly.

Iain just reported a problem with LEDs which may be related..  In that case the forward drop larger, and so the reverse current is so small that it's creating numerical errors.

Felthry, 2018-10-30:
Just watch what happens when you set the Vf to something like 26mV ([yes, these exist](https://www.digikey.com/product-detail/en/texas-instruments/SM74611KTTR/296-35688-1-ND/3911155)). (they're actually FETs with driving circuitry to make them act like nearly-ideal diodes, but still.)

DerKarhu, 2018-10-31:
Kind of an aside to this conversation, I wondered if there is, or could be, any way to map SPICE models into a diode model here?

pfalstad, 2018-10-31:
Working on it

pfalstad, 2018-11-02:
ok now you can use spice-like models for diodes.  it fixes the problems with schottky diodes and the issue Iain reported with the LEDS.  Give it a try.

BrandonLWhite, 2018-11-02:
@pfalstad Where can I find the commit or build with your fix?

Felthry, 2018-11-02:
@BrandonLWhite It's at falstad.com/circuit.
@pfalstad I tried it, and it does work well, but the interface is clunky, and difficult if you want to just input values from a datasheet. I would have thought having an I_s and V_f would be sufficient, then just fit the exponential to those two points?

BrandonLWhite, 2018-11-02:
Ah thank you.  I selected the 1N5711 Schottky option available now for the diode, and the above circuit behaves as expected.  Awesome!

sharpie7, 2018-11-02:
I have just pulled the update in to this version.

DerKarhu, 2018-11-02:
Hi there;

Long post, but, for anyone interested, I found this excel spread sheet that does a calculation for Is and n, leaving the R value for us to plug in, from looking at the I/V curve roll-off...

from this discussion, "W5V0' answered (there are others, later, but this seems to work well-enough), and linked to this spreadsheet

http://www.tyblu.ca/misc/diode_modeling.xlsx

and, more, LibreOffice calculations by 'Roland', in the last answer...

 but it is easy to solve the Shockley diode equation algebraicly. Just note that the "minus 1" in the formula is very irrelevant for forward currents that are an order of magnitude greater than Is, which is very small, say, 1E-12 A. Find just two points in the graph with easy to read I and V values, and plug these into the formula. Dividing both formula's eliminates Is, so N is easy to compute. Then fill in N in a formula to find Is.
Here are my LibreOffice Calc macro's in Basic:
Const Q as double = 1.6E-19
Const K as double = 1.38E-22
Const T as double = 300

rem The Shockley diode equation, to build the graph Id(Vd) for hardcoded values of Is and N
Function shockley(Vd as double) as double
    Const Is1 as double = 5.94463E-18
    rem Note that 'Is' is a reserved word and cannot be the name of a variable
    Const N as double = 0.191367
    shockley = Is1 * (exp(Vd * Q / (N * K * T )) - 1)
End Function

rem Step 1 in solving the diode equation for N using values from a graph
Function ComputeN(V1 as double, V2 as double, I1 as double, I2 as double) as double
    ComputeN = (Q / (K * T)) * (V1 - V2) / (log(I1) - log(I2))
End Function

rem Step 2 in solving the diode equation for Is
Function ComputeIS(V as double, I as double, N as double) as double
    ComputeIS = I / (exp(Q * V / (N * K * T)))  
End Function

rem for debugging
sub Test
    dim N as double
    N = ComputeN(1.85, 1.3, 0.1, 1.5E-6)
    dim Is1 as double
    Is1 = ComputeIs(1.85, 0.1, N)
end sub

If you look at the formula's, you might recognize simply the description of a straight line with slope of q/NkT but also of delta Log(Id) / delta Vd.
I get a somewhat similar value for Is: 5.94E-18 = 5.94 atto-ampere (W5VO found 1 aA), but much different N = 0.19 (W5VO found 1.8, typo?), still the data also computes back to the same graph:

The column Vd are the voltages, Id is the diode current according to the real formula, Id0 is the current with the simplified formula where the "minus 1" is changed to "minus zero". As Id0 is a true exponential curve, you can take the logarithm in column Id0_log. (You cannot take the log of a curve that becomes zero and negative like Id) The plot is from Id0_Log versus Vd. In this plot I made the lowest part dotted, because there it is NOT the actual diode current anymore, but does show the value of Is at the intersection with the Y-axis.
Following the exponential curve to the left brings you asymptotically to zero. But the "minus 1" subtracts an amount of Is, so that the real diode curve goes through the origin and, with negative voltages, shows a reverse leakage current of amount Is.
If the original manufacturers curve would have been on a really large log plot, we could have simply used a ruler to extend the straight line downwards to easily find Is at Vd=0 and then compute N, instead of compute first N then Is with the above macro's. The ruler method has been described in "The Spice Book" by Andrei Vladimirescu (1994).

Felthry ---02.11.2018 13:56:38---@BrandonLWhite It's at falstad.com/circuit. @pfalstad I tried it, and it does work well, but the intFelthry ---02.11.2018 13:56:38---@BrandonLWhite It's at falstad.com/circuit. @pfalstad I tried it, and it does work well, but the int

pfalstad, 2018-11-03:
We've tinkered with our own "simplified" model over time, and it's caused problems.  I wanted to go with a proven model.  There are a lot of existing models out there for spice, and a fair amount of documentation on how to develop your own.  Also, the parameters in this model are exactly what we need for our simulation.  Before, we were converting internally to this format, and not doing a great job.  (Also there was no way to simulate series resistance before.)

It might be good to have a dialog to input the diode parameters I_s, V_f, and current at V_f, and have it calculate the emission coefficent.  Or maybe that could be a separate javascript worksheet since it's not something you do very often.

DerKarhu, 2018-11-05:
Thanks for all of your work on this, and other, simulations... useful, and fun, too ;-)

Der Karhu

Paul Falstad ---03.11.2018 21:46:58---We've tinkered with our own "simplified" model over time, and it's caused problems.  I wanted to goPaul Falstad ---03.11.2018 21:46:58---We've tinkered with our own "simplified" model over time, and it's caused problems.  I wanted to go

DerKarhu, 2018-11-05:
Hi Paul;

I noted that the default LED appears to be a 'blue LED', which is usually the case for 'white' LEDs, as well... 
Best regards,

Der Karhu

Paul Falstad ---02.11.2018 02:27:18---ok now you can use spice-like models for diodes.  it fixes the problems with schottky diodes and thePaul Falstad ---02.11.2018 02:27:18---ok now you can use spice-like models for diodes.  it fixes the problems with schottky diodes and the

sharpie7, 2019-03-06:
Believe fixed.

closed 2019-03-06 by sharpie7