sharpie7/circuitjs1 #172: Capacitor In Parallel Orientation

DrDean006, 2017-10-03:
When placing a capacitor in parallel orientation Amperes to the capacitors size (Farads) is always in the ratio of 400 : 1 - independently of voltage. Is that correct, and if it is would someone be able to post the formula that gives the ratio 400 : 1 in respect of the above?

pfalstad, 2017-10-03:
hi, could you give an example circuit with File->Export as Link or File->Export as Text?

Are you talking about a circuit like this:  http://www.falstad.com/circuit/e-cappar.html (on the left)

DrDean006, 2017-10-04:
If you increase the size of the capacitor (attached circuit) amperes will increase in proportion (100 uf = 40ma at capacitor / 10 uf = 4ma at capacitor).
[circuit-20171004-1518.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/1354392/circuit-20171004-1518.circuitjs.txt)

pfalstad, 2017-10-04:
The behavior you're seeing is not realistic.  You should have a resistor in series with the capacitor:

http://tinyurl.com/y7aaemlr

sharpie7, 2017-10-09:
@pfalstad Agreed, but, unlike some other examples, the original circuit is analysable and has non-singular solutions. Is there a feature in the sim's model that causes it to create a current through a capacitor even if its ends are short-circuited by a wire?

pfalstad, 2017-10-09:
The capacitor model consists of a current source in parallel with a resistor.  If you're using the trapezoidal approximation, the value used for the current source depends on the voltage across the capacitor and the current in the last iteration.  If the capacitor is shorted, the voltage will be zero.  But the current could still be nonzero if the capacitor used to have voltage across it, and the circuit was not reset.  In that case the current across the capacitor will oscillate back and forth.

If the backward euler approximation is used (trapezoidal is switched off), then the current source value only depends on the voltage, so shorted capacitors will always have zero current.  The trapezoidal approximation is generally more accurate, but has more of these sorts of issues with instability when things change suddenly.

We used to have code in there to reset the capacitors when they get shorted, but it doesn't seem to be working at the moment.

closed 2017-10-21 by sharpie7