sharpie7/circuitjs1 #174: NE555 sawtooth generator : frequency display error

Philexium, 2017-10-11:
Hi,

When I change the value of the capacitor (200nF) to lower it, the new frequency is no more displayed on the left hand side (Output).

Any idea ?

Thank you !

Philexium, 2017-10-11:
For information :
I have tried with 3 browsers IE, Firefox and Chrome.

The problem is the same.

pfalstad, 2017-10-11:
Seems to work for me.  You just need to wait until the frequency changed has filled the whole width of the scope.  Speeding up the simulation will help with that.  What did you set the capacitor value to?

Philexium, 2017-10-11:
Hi Paul
Thank you for your message.
I want the NE 555 to generate a sawtooth at 100 Khz.
I have tried to set the capa at 2nF

Philexium, 2017-10-11:
This is a printscreen.
As you can see no more Hz value displayed under the line "output"
[image]

pfalstad, 2017-10-11:
Hi,

ok if you want to simulate a frequency that high, you should probably decrease the time step.  go to the menu and do Options->Other Options... and set the time step size to 200n.

you could also increase the speed of the scope (by right-clicking on the scope and selecting "Speed 2x" multiple times, or using the mouse wheel) but the simulation will not be as accurate.

pfalstad, 2017-10-11:
If you want 100khz set the cap to 100p and the time step to 5n.

closed 2017-10-21 by sharpie7