sharpie7/circuitjs1 #176: Error in Voltage Drop in Parallel Resistors

sjhillier, 2017-10-20:
I drew up this simple series parallel circuit with a 3V power supply (see picture) the problem is that the voltage drop is reading 1.06V over a single 33 ohm resistor when it should have been 1.06V over all three 33 ohm resistors and 353.3mV over each one. Kind of a problem since it means that my total voltage drop was somehow more than the power supply was even putting into the circuit.

[image]

pfalstad, 2017-10-20:
looks right to me..  The resistors are in parallel, so you don't add the voltage drops.  Total voltage drop is .97V (for one 10 ohm resistor) + .97V (for the other one) + 1.06V = 3V

sjhillier, 2017-10-20:
No I'm afraid that is incorrect. It should should 1.06V if all three resistors in parallel were selected, they were not. The simulation is supposed to simulate real life circuits no? If you were to physically measure the voltage drop over each of those 33 ohm resistors you would measure 353.3mV not 1.06V.

colinhowell, 2017-10-20:
@sjhillier

The simulator is correct. You're misunderstanding how voltage works with parallel circuit elements and need to study the subject more carefully.

The voltage does **not** get split amongst the three parallel 33-ohm resistors. Rather, all three resistors see the **same** voltage drop, and the **current** gets evenly split among the resistors. The three parallel resistors pass the same **total** current as a single 11-ohm resistor would, giving the voltage drop you see.

The voltage **is** divided among the three **series** resistances: the two 10-ohm resistors and the 11-ohm combination of the three parallel 33-ohm resistors.

Read up more on [series and parallel circuits](https://en.wikipedia.org/wiki/Series_and_parallel_circuits) and on [Kirchhoff's circuit laws](https://en.wikipedia.org/wiki/Kirchhoff's_circuit_laws): these are extremely fundamental to the operation of electric circuits.

If you think of the circuit as a water path, with the voltage source being a pump, each resistor being a tilted pipe whose cross-sectional area corresponds to its resistance, and the voltage drop across each resistor being the difference in height between the two ends of that pipe, this behavior may make a bit more sense. The so-called "[water analogy](https://en.wikipedia.org/wiki/Hydraulic_analogy)" can sometimes be pretty useful in understanding basic electric circuit behavior, though you do have to remember that it has limits.

closed 2017-10-20 by sjhillier