sharpie7/circuitjs1 #180: Potential lead for MOSFET Convergence Errors

OmegaHaxors, 2017-11-08:
http://tinyurl.com/y7ffn254

Creating this small circuit causes the MOSFET to ring chaotically when the control voltage is 0. On its own, it's not a harmful bug but I find that too many of these going off at once creates a Convergence Failure.

The example provided is extremely idealistic with its 1M beta and 0 Threshold Voltage, though I think it pinpoints the underlying issue that has been creating countless Convergence errors for me in the past.

aheid, 2018-01-16:
If you replace the MOSFET with a 800M resistor (which is what the MOSFET presents itself as when in the off state), you also get oscillations. The oscillations go away when you reduce the timestep to 100n or less even with the MOSFET.

So this seems to be just a case of the CFL condition or something analogous.

E3V3A, 2019-02-25:
> The oscillations go away when you reduce the timestep to 100n or less even with the MOSFET.

@aheid 
@sharpie7 
I would be good to have a (Github Repo) Wiki page about how to correctly simulate MOSFET's. 
(How is anyone to know that we need to change the timestep to get correct results?)

aheid, 2019-02-25:
@E3V3A As I remember it, the core issue is that very large resistors (and similar) can lead to a very "stiff" system, and the stiffer the system the smaller timestep one need to use. Ideally the simulator should be able to calculate some timestep bounds, ala CFL[1], and guide the user to select a maximum viable timestep.

Alternatively I remember wrong and the above is nonsense :)

Until that is in place, it would indeed be nice to document this.

[1]: https://en.wikipedia.org/wiki/Courant%E2%80%93Friedrichs%E2%80%93Lewy_condition

pfalstad, 2019-02-26:
You should turn off the trapezoidal approximation checkbox on any inductor attached to a switch, or something that acts like one.  This should be documented.  It might be good for the simulator to detect this oscillation and warn about it, maybe providing a link the documentation.