sharpie7/circuitjs1 #183: Can we have a battery?

Benur21, 2017-12-01:
A battery which stores energy, and then we could move it to another components and see how its charge decreases over time, etc...

PlanMan1717, 2020-05-26:
__A capacitor__ is maybe what you're looking for. This may be infeasible, as the approx. storage capacity for a AA battery is around 6000F charged to 1.5V--- and if you were to charge it (realistically), it would take forever.

__The quick method__ is to take a 10uF cap and charge it to 1.5V or whatever, *then* change its capacitance to 6000. It will retain the charged voltage, and has a wholly capacitance.

Benur21, 2020-05-26:
> **A capacitor** is maybe what you're looking for. This may be infeasible, as the approx. storage capacity for a AA battery is around 6000F charged to 1.5V--- and if you were to charge it [...]

Tell me if I understood it. A 6000F capacitor behaves exactly like a AA battery? Or is that only the same when discharging?