sharpie7/circuitjs1 #187: Translation

SebCanet, 2017-12-09:
Hello,
I like this project and wanted to use it for my students, but they are french and not very ease with english. How can I translate it ?
Could you enable a multi lang capabilities in Circuit ?
Thanks.

sharpie7, 2017-12-09:
Bonjour!

The application does support localization. I haven't tried it myself, but @pfalstad knows how it works.

I think you just have to create a file similar to the locale_de.txt (but called locale_fr.txt) that is in the directory /src/com/lushprojects/circuitjs1/public/

SebCanet, 2017-12-09:
Thanks a lot !

closed 2017-12-09 by SebCanet

pfalstad, 2017-12-09:
That's right, just translate that file and send it to me and I can help incorporate it into the site.

SebCanet, 2017-12-09:
And the real will be to install Eclipse, understand Eclipse, anger and shout on Eclipse, try to install GWT, anger and shout on Eclipse, try to compile, anger and shout on Eclipse...
I'm not ease at all with this paritcular environnment.
Could it be possible for 'eclipse newbie' to have relkeases on Github ?
Thanks.

pfalstad, 2017-12-09:
You don’t need to bother with eclipse, if you just want a French version of the simulator.  Just translate that file and send it to me.