sharpie7/circuitjs1 #193: [Feature Request] Can we have an optocoupler?

AC-Model, 2018-01-07:
For my developments it's very useful to have an optocoupler (something like Typ: CNY17-1-000E).
If it is possible, give me your support please.

Thank you

bohix, 2018-06-20:
Same request please, or explain how to reproduce one with available components.
In my case models 6N137 and SFH620A.
Thank you!

sharpie7, 2018-08-26:
It's a reasonable request to add an optocoupler to the project. Though, as I have said elsewhere, this project generally doesn't model particular components, but rather generic representations of components.

The problem is that I have never seen any general mathmatical model for the behaviour of an optocoupler. If someone knows one then please add a link to this discussion.

ibelectronics, 2018-12-14:
It would be fine to add general optocoupler model (like PC817) to have galvanic isolation between input and output. Mr.Falstad, please add it into the circuit simulator. Thanks a lot.

E3V3A, 2019-02-26:
I agree, optocouplers are surprisingly popular. I think they are even present in some Apple chargers.

closed 2019-05-01 by sharpie7