sharpie7/circuitjs1 #199: Bug - Convergence failure not reported

aheid, 2018-01-16:
The following circuit does not converge properly most of the time. Hitting reset multiple times (10-20 times) may get it to converge and then it'll work fine.

````
$ 1 1e-12 10.20027730826997 50 5 50
R 160 176 112 176 0 0 40 0.618 0 0 0.5
r 480 32 480 128 0 1
f 416 192 480 192 0 4 1.19
w 480 176 480 128 0
g 480 336 480 368 0
370 480 208 480 336 1 0
a 208 160 304 160 0 9 0 1000000 0.00009189802221953869 0.618
w 304 160 416 160 0
w 416 192 416 160 0
a 416 80 320 80 1 9 0 1000000 9.09090908999011 9.09090909090909
r 480 32 416 32 0 100000
r 480 128 416 128 0 100000
r 416 128 320 128 0 1000000
w 320 128 320 80 0
w 416 128 416 96 0
w 416 64 416 32 0
r 416 32 320 32 0 1000000
g 320 32 320 48 0
w 320 80 208 80 0
w 208 144 208 80 0
w 208 176 160 176 0
368 208 80 160 80 0 0
368 160 176 160 112 0 0
368 416 192 368 192 0 0
370 560 32 480 32 1 0
R 560 32 592 32 0 0 40 10 0 0 0.5
o 2 64 0 4099 10 25.6 0 2 2 3
o 1 64 0 4099 0.00244140625 0.003125 1 2 1 3
````

What I find especially weird is how it can show almost 15A going through the MOSFET yet almost nothing through the shunt resistor, all this without telling me it failed to converge. At the very least it should report failure.

Is there really no check or enforcement of the energy balance? Ie, current into point + current out of point = 0? I'm not at all familiar with the method used for the sim[1], but isn't there some way like the [projection method](https://en.wikipedia.org/wiki/Projection_method_(fluid_dynamics)) to enforce the energy balance?

In any case, thanks for you work on this excellent tool!

[1]: a reference would be much appreciated!

E3V3A, 2019-02-26:
Yes, we'd love to see some more material and have some references to how the simulation is done, and what methods it is using.

pfalstad, 2021-12-11:
This seems to be fixed now.

closed 2021-12-11 by pfalstad