sharpie7/circuitjs1 #203: Bug - 555 Timer Circuit Current

rrbman, 2018-02-03:
The input current at  Vcc does not take into account the output current. If a load like 1kOhm is put at the output the input current does not change. For the 10 V input, he output current becomes 10mAmps while the input current stays at 670uAmps. This causes an error if there is a resistive element in series with the input Voltage since the voltage drop to Vcc will be too low..
It looks like the OpAmp output current is not figured into the input current.Load.
Input the file for 555 timer into the simulation. Add a  load to ground like 1kOhm to the Output. Examine in and out current. It should not put out more than it takes in.

sharpie7, 2018-08-26:
Yes, this does seem to be a problem. The 555 simulation uses a voltage source to simulate the output, and this pulls in current externally to the simulated circuit which isn't correctly allowed for at the Vin. @pfalstad may know how best to fix this.

pfalstad, 2018-08-26:
I’m not sure why I did it that way.  It probably wouldn’t be that hard to fix.

closed 2019-03-06 by sharpie7