sharpie7/circuitjs1 #207: Feature: Pan and zoom Meta key (add Ctrl support)

babs, 2018-02-25:
It would be nice to have on top of Alt, Ctrl to also triggers the pan/zoom.
(or making it parametric (shit ? combinaison of several (alt + shift or ctrl + shift) ?))

Reason: Alt + click is used by some window managers to drag or move window around, and Alt + mousewheel to zoom for accessibility purposes, while control + click and control + mouse wheel are free of bindings.

babs, 2018-02-25:
or middle click drag ?

pfalstad, 2018-05-26:
control click brings up a context menu on mac.  but I can do middle click drag.

I don't understand why you're mentioning alt-mousewheel here, since the mousewheel adjusts the zoom by itself with no modifiers.

babs, 2018-05-29:
Middle click alone doesn't work for pan on XFCE (maybe linux in general).

pfalstad, 2018-05-29:
I meant, I will add the ability to drag with middle mouse.

babs, 2018-05-30:
Sorry I misunderstood. thanks :-)

pfalstad, 2019-03-06:
added in 3f9d1d28df5980b0ab5d8b5a70e4e8ec861f1bf3

closed 2019-03-06 by sharpie7