sharpie7/circuitjs1 #212: Realtime Simulation?

tinuxman, 2018-03-04:
I was just wondering if it's possible to simulate in real time? I built a switch-on delay and would like to test the "real" seconds. Thanks.

NPG-3466, 2018-03-06:
maybe a changeable simulation speed.

NPG-3466, 2018-03-06:
I just found out that if you do that ac sources won't work properly anymore. their frequency will change to 0.125 Hz

sharpie7, 2018-03-06:
As you have found, the timestep needs to be small compared to the period of the highest frequency in the circuit of interest. That is how the simulation works: by looking at how the circuit changes over very small increments of time and summing these up to produce the long term behaviour.

Realtime simulation is no problem. You just need a fast enough computer. Good luck finding one!

closed 2018-03-06 by sharpie7