sharpie7/circuitjs1 #213: Re-Coding the circuit simulator in HTML5

VCAPPhysics, 2018-03-07:
Hi there,

I find the simulator extremely useful. Have you thought of porting it from Javascript to HTML5. I use Chromebooks and Javascript doesn't work.

Cheers,

Juan Pineda

mmiscool, 2018-03-07:
The following link works in google chrome.
http://www.falstad.com/circuit/

sharpie7, 2018-03-08:
Javascript IS the technology used by HTML5 for client-side web applications. The simulator needs to run client side to provide the smooth interactivity and to make the hosting economically viable. 

I don't have a Chromebook to test with, but the application does work with Chrome and all other recent browsers so it should also work on Chromebooks unless you have a configuration (e.g. a Javascript blocker) preventing it.

closed 2018-03-08 by sharpie7

VCAPPhysics, 2018-03-08:
ok. I will check it.

Thanks,

Juan

*Juan Pineda**MASc. (Power Systems), B.Eng. (Electrical), B.Ed. (Science)*

Science Teacher (Physics and Chemistry), Science Department Head, IT
Coordinator, and Varsity Soccer Head Coach

jpineda@mail.vc.bc.ca <ccousineau@mail.vc.bc.ca>

604-261-4285 <(604)%20261-4285> Telephone

604-261-2284 <(604)%20261-2284> Fax

*Vancouver College *

5400 Cartier Street

Vancouver, BC V6M 3A5

www.vc.bc.ca

<http://www.vc.bc.ca/>   <http://www.ournextcentury.ca/>

*CONTINUE THE LEGACY. SUPPORT OUR NEXT CENTURY
<http://www.ournextcentury.ca/>.*

pfalstad, 2018-03-08:
It works on my daughter's chromebook.

VCAPPhysics, 2018-03-08:
Excellent! It works on all kinds of mobile devices. This is a great tool
for my AP Physics students.

Thank you,

Juan Pineda