sharpie7/circuitjs1 #215: Language selection

Pivetta, 2018-04-01:
I have the feeling I am missing something.
Am now guest on another computer with language settings in German, and am not allowed to change them. I get now access to the simulator with German localization which makes it about unusable to me.
I can`t find in the settings anything to change the simulator language.
I´d apreciate such menu feature.

pfalstad, 2018-06-13:
Try this link: http://www.falstad.com/circuit/circuitjs.html?lang=en

dsvf, 2018-06-20:
This is a suitable interim solution, but it would be great to either have language selection as a menu option/"Other options" item, or at least optionally remember the language setting (though that would probably require cookies, and it seems that you do not use them).

caracole, 2019-03-23:
I am of the opinion of dsf. The automatic translation by google (google chrome) is imperfect and slowed considerably the application.

pfalstad, 2019-03-23:
ok I will add an option to set this from the Other Options dialog and have it remembered.

pfalstad, 2020-09-10:
Added

closed 2020-09-10 by pfalstad