sharpie7/circuitjs1 #219: j/k flipflops naming "issue"

BrayanKhosravian, 2018-05-16:
first of all i have to say that this is a very great software TBH. its not complicated at all. its able to drag and drop elements, wire them and get a output of ea element without inserting weird inputs.

thats the reason why i post a issue, because i like this software a lot so i could maybe help u to improve it.

well i just figzred out that the j/k flip flops are inverted so they should maybe be called "inverted j/k flip flop" or for a short version "inv. j/k flipflop"

sharpie7, 2018-05-19:
Thanks for your support.

The JK flipflops look correct to me. In what way do you think they are inverted?

BrayanKhosravian, 2018-05-19:
The clock input is inverted with a " ° " sign

pfalstad, 2018-05-25:
It seems like there are both positive and negative edge triggered JK flip flops out there.  I can make the trigger behavior an option.

BrayanKhosravian, 2018-05-26:
Oh nice i didnt know about that ty

closed 2018-05-26 by BrayanKhosravian