sharpie7/circuitjs1 #221: Feature request: Nonideal op amp

Felthry, 2018-05-19:
At the risk of making this too detailed, it would be nice to have access to more parameters in the op amp model. It already has finite open-loop gain, but that open-loop gain isn't user-controllable. Modelling of input offset voltage and input bias current might also be nice to have, but probably too esoteric to be worth adding; it seems like one goal of this (correct me if I'm wrong) is to be simple and easy to understand, and adding in more nonideal behavior might compromise that.

To be clear, the main feature I'm requesting here is being able to set the open-loop gain of an op amp.

her6betz, 2018-06-01:
I would like to support the voltage pin request on the op-amp. I got a setup here with an LM324 using +O---9V---O-ground-O---9V---O- as voltage source. However there seems no possibility to connect this to your op-amp-simulation.

sharpie7, 2018-08-26:
Right-click an op amp and choose "EDIT" - you can set the power supply voltages and the gain.

I agree @Felthry that if you want to model things like offset voltage and input bias current that other tools, like SPICE, would be more appropriate. We generally don't to go a lot of detail in this project.

closed 2018-08-26 by sharpie7