sharpie7/circuitjs1 #226: Steady-state DC operating point button

endolith, 2018-05-31:
When simulating audio circuits with realistic frequencies and DC offsets, the offsets can take a _looooong_ time to settle.  I often modify the circuit to charge up the capacitors quickly, then modify it back, run the simulation for a while at maximum speed, etc.  

It would be nice if there were a button that would calculate the steady-state operating point of the circuit, instantly set all DC voltages to those values, and then continue simulating.

pfalstad, 2018-05-31:
hi, not a bad idea.  I might make it a menu item rather than a button.  Do you have some example circuits you could send me?

sharpie7, 2018-05-31:
@pfalstad when I looked at this a while back I concluded that the visualisation wasn't a major cause of delay in the program, so if you want to significant;y speed up the convergence to the DC steady-state you would have either use a new algorithm or change the time-step per iteration.

@endolith Don't forget that you can use the simulation speed slider to speed up the simulation while it calculates bits that you aren't interested in viewing in detail.

pfalstad, 2018-05-31:
What I had in mind was a button that would replace all the capacitors with large resistors, all the AC sources with DC sources, etc. run the circuit for one iteration, and then continue at that point.  That seemed to work for the few examples I tried but I don't know if they're the same type of circuits that endolith is working with.

endolith, 2018-06-05:
Here's an example circuit: http://tinyurl.com/y9cz7sfq

pfalstad, 2018-06-05:
Great, it works fine on that circuit.

pfalstad, 2018-06-13:
Ok try File->Find DC Operating Point.

endolith, 2018-06-20:
Seems to work fine

closed 2018-08-26 by sharpie7