sharpie7/circuitjs1 #227: op amp simulation without voltage pins

her6betz, 2018-06-01:
I would like to support the voltage pin request on the op-amp. I got a setup here with an LM324 using +O---9V---O-ground-O---9V---O- as voltage source. However there seems no possibility to connect this to your op-amp-simulation.

pfalstad, 2018-06-01:
Hi, if the + and - voltages are fixed like in this example, you can just right click, Edit, change the max and min voltages to +9 and -9.

her6betz, 2018-06-01:
Thanks, but I cannot see, how to connect the voltage source to the op-amp and so make a run possible?

billcollis, 2018-06-01:
I made a model with power pins to show students that opamps weren't magic

OpAmpSwapPwrElm.java
`package com.lushprojects.circuitjs1.client;

//import java.awt.*;
//import java.util.StringTokenizer;

/* Bill Collis June 2015
 * 
 * want to show current to/from power supply connections
 * 
 * ID = 352
 * */

    class OpAmpPwrElm extends CircuitElm {
	int opsize, opheight, opwidth, opaddtext;
	double maxOut, minOut, gain, gbw;
	boolean reset;
	final int FLAG_SWAP = 1;
	final int FLAG_SMALL = 2;
	final int FLAG_LOWGAIN = 4;
	public OpAmpPwrElm(int xx, int yy) {
	    super(xx, yy);
	    noDiagonal = true;
	    maxOut = 15;
	    minOut = -15;
	    gbw = 1e6;
	    setSize(sim.smallGridCheckItem.getState() ? 1 : 2);
	    setGain();
	    compLabel="U"; //BC compID will be generated in CirSim.java
	}
	public OpAmpPwrElm(int xa, int ya, int xb, int yb, int f,
			StringTokenizer st) {
	    super(xa, ya, xb, yb, f);
	    maxOut = 15;
	    minOut = -15;
	    // GBW has no effect in this version of the simulator, but we
	    // retain it to keep the file format the same
	    gbw = 1e6;
	    try {
		maxOut = new Double(st.nextToken()).doubleValue();
		minOut = new Double(st.nextToken()).doubleValue();
		gbw = new Double(st.nextToken()).doubleValue();
	    } catch (Exception e) {
	    }
	    noDiagonal = true;
	    setSize((f & FLAG_SMALL) != 0 ? 1 : 2);
	    setGain();
	    compLabel="U"; //BC
	    compID=0;
	    if (st.hasMoreTokens()){
		    String tok = st.nextToken(); 
		    if (tok.indexOf('@')>-1)
		    	compID = new Integer (tok.replace('@', ' ').trim()).intValue();
	    }

	}
	void setGain() {
	    // gain of 100000 breaks e-amp-dfdx.txt
	    // gain was 1000, but it broke amp-schmitt.txt
	    gain = ((flags & FLAG_LOWGAIN) != 0) ? 1000 : 100000;
	    
	}
	String dump() {
		if (compID>0)//BC
			return super.dump() + " " + maxOut + " " + minOut + " " + gbw + " @" + compID;//BC
		else
			return super.dump() + " " + maxOut + " " + minOut + " " + gbw;
	}
[... 157 lines cut ...]
		maxOut = ei.value;
	    if (n == 1)
		minOut = ei.value;
	    if (n == 2)
	    	compID = (int)ei.value;
	}
	@Override double getCurrentIntoPoint(int xa, int ya) { 		
	    if (xa == x2 && ya == y2)		
		return -current;		
	    return 0;		
	}
}
`

OpAmpSwapPwrElm.java
`package com.lushprojects.circuitjs1.client;

class OpAmpSwapPwrElm extends OpAmpPwrElm {
	public OpAmpSwapPwrElm(int xx, int yy) {
	    super(xx, yy);
	    flags |= FLAG_SWAP;
	}
	Class getDumpClass() { return OpAmpPwrElm.class; }
    }
`

pfalstad, 2018-06-09:
Sorry for the late response her6betz, I didn't see your latest message for some reason.

I got your circuit working.  The problem is that the time step was way too small to show anything happening in the circuit in a reasonable amount of time.

http://tinyurl.com/ycb5tg55

The op amps in the simulator are always connected to an external voltage, but this isn't shown, just to simplify the view of the circuit.

pfalstad, 2018-06-09:
Thanks billcollis, can I use this?

Exaeta, 2018-09-10:
Hello. I wanted to use one of these to test a voltage regulator design. But it wont be a very good test if the voltage is already regulated.

Would it be possible to add these pins?

ormaaj, 2019-03-25:
If the CCVS expression accepted a function that takes an input name and returns its voltage then it might be possible to have supply pins that sink current correctly. Still circuits that have multiple sources are trouble because there's no way to assert what proportion of current at an input is supplied by which source and ensure the output equals the sum of the input currents at all times.

Also the op amps have no option to choose DC offset like the sources do (and those have some weird behaviours). The only way I've found to eliminate side effects from magic sources is to ensure only 2-termanal sources are used and make sure nothing generates negative voltages. Sometimes the mere presence of the ground symbol or a 1-terminal source alters the circuit, even if it's a floating node.

pfalstad, 2020-12-13:
You can use Op Amp (real) now.

closed 2020-12-13 by pfalstad