sharpie7/circuitjs1 #228: Optocoupler and output function not in menu

her6betz, 2018-06-03:
I was looking for an optocoupler (like issue #193), but found none in the menu. In one of the comments I came accross an "output function", also not in the menu ...

1) Why can I not find some optocoupler in the menu?

2) What is an output fct.? Probably simply a black box? I found, a value of "-0.2*i" in this device worked (see image below), but "0.2*i" did not so well?

3) Could I mimic an optocoupler using an output function?

output_fct.txt:
$ 1 0.000005 10.20027730826997 50 5 43
215 0 128 48 128 0 2 -0.2*i

[image]

pfalstad, 2018-06-03:
Hi, that's a current-controlled current source (CCCS), and it's in the menu.  "Output function" isn't being translated to German, which is a bug.  The output function is just an expression for the output current in terms of the input current.  Why doesn't 0.2i work?  It seems to work for me.

her6betz, 2018-06-03:
Thanks for the fast reply. I'm a hobbyist and very enthusiastic about your simulator and your support ...

So obviously CCCS has nothing to do with an optocoupler ...

sharpie7, 2018-08-26:
Correct - a CCCS is nothing to do with an optocoupler. I will close this issue as #193 already contains a request to add an optocoupler.

closed 2018-08-26 by sharpie7