sharpie7/circuitjs1 #229: Inserting testpoint changes toggling behaviour

her6betz, 2018-06-06:
In the appended circuit of a T-Flipflop, the toggling behaviour changes, if I insert a testpoint at one of the transistor collectors. I have to reload the file, to get to the usual behaviour of the leds, when not  inserting the testpoint. A bug?

[circuit-_2transTFlop_pulsed.txt](https://github.com/sharpie7/circuitjs1/files/2075257/circuit-_2transTFlop_pulsed.txt)

pfalstad, 2018-06-07:
Hi, where are you inserting a testpoint?  How does the behavior change?

You say that reloading the file fixes it?  maybe you could send a screenshot...

her6betz, 2018-06-07:
I load the file above "circuit-_2transTFlop_pulsed.txt" into your simulator. Running it, I can toggle the leds by actuating the switch several times.

Now I insert a testpoint at one of the transistors collector. I then cannot toggle the leds any more ... even after removing the testpoint again.

From the screenshot:
[image]

pfalstad, 2018-06-07:
Are you talking about the situation where both LEDs are lit up?  I can get that to happen by resetting the circuit, without a test point.  That is probably related to the circuit being too symmetric.  If you change one of the 470 ohm resistors to 480, that will go away.

Other than that, I'm not seeing any problem with adding a test point.  I'm not sure how you got the situation above with the test point reading 0 V and no voltage or current anywhere in the circuit.  Maybe that's after clicking Reset?

closed 2018-08-26 by sharpie7