sharpie7/circuitjs1 #231: How do you make an inductor with a derivative in the middle, without using two inductors?

carlosdelfino, 2018-06-13:
I am studying the circuit of a ZVS oscillator and need to simulate an inductor of 4uH, but derived in the middle, similarly to what is in the diagram described below.

```
$ 1 0.000005 2.803162489452614 30 5 50
f 304 176 352 176 0 2 0.02
f 304 304 352 304 8 2 0.02
g 224 240 224 272 0
c 416 160 416 320 0 6.799999999999999e-12 0.000004412264818398626
R 528 48 528 16 0 0 40 24 0 0 0.5
l 528 80 528 144 0 0.00019999999999999998 2.016366778784425
w 528 48 528 80 0
w 416 160 480 192 0
w 416 320 480 288 0
w 416 160 384 160 0
w 416 320 352 320 0
w 384 240 384 288 0
w 368 128 368 160 0
w 208 128 208 304 0
w 256 304 304 304 0
w 368 352 368 320 0
w 528 240 528 144 0
d 272 128 368 128 1 0.805904783
r 176 176 112 176 0 470
r 176 304 112 304 0 470
w 208 304 256 304 0
w 192 176 256 176 0
w 112 304 112 176 0
w 112 176 112 80 0
w 528 80 112 80 0
d 192 352 368 352 1 0.805904783
w 352 160 368 160 0
w 352 320 368 320 0
w 208 128 272 128 0
w 304 240 384 240 0
z 304 240 304 304 1 0.805904783 12
r 256 176 256 240 0 10000
r 256 240 256 304 0 10000
w 304 240 256 240 0
w 176 304 208 304 0
w 384 192 384 240 0
w 192 176 192 352 0
w 176 176 192 176 0
z 304 240 304 176 1 0.805904783 12
w 256 176 304 176 0
d 384 288 384 320 1 0.805904783
d 384 192 384 160 1 0.805904783
w 368 320 384 320 0
w 368 160 384 160 0
w 352 288 384 288 0
w 352 192 384 192 0
w 224 240 256 240 0
l 480 192 480 240 0 0.000002 -1.008183379077077
l 480 240 480 288 0 0.000002 1.0081833997073388
w 528 240 480 240 0
o 14 64 0 4099 20 0.025 0 4 14 3 39 0 39 3
o 8 64 0 4107 40 6.4 1 2 8 3

```

pfalstad, 2018-06-13:
You could use a tapped transformer with nothing connected to the other side.

I'm trying to get this circuit to work but I'm not having much luck...

carlosdelfino, 2018-06-14:
I did the project [using a tapped transformer](http://carlosdelfino.eti.br/circuitjs1/?startCircuit=zvs-inductor.txt) and it did not work correctly.

The Code is below:

```
$ 1 0.000005 2.803162489452614 30 5 50
f 304 176 352 176 0 2 0.02
f 304 304 352 304 8 2 0.02
g 224 240 224 272 0
c 416 160 416 320 0 0.0006799999999999999 0.0004080582445027403
R 448 48 432 16 0 0 40 24 0 0 0.5
l 448 80 448 144 0 0.00019999999999999998 2.0163669184362627
w 448 48 448 80 0
w 416 160 448 208 0
w 416 320 448 272 0
w 416 160 384 160 0
w 416 320 352 320 0
w 384 240 384 288 0
w 368 128 368 160 0
w 208 128 208 304 0
w 256 304 304 304 0
w 368 352 368 320 0
169 528 272 448 272 0 0.0003 20 3.5912502897687826e-17 -1.008252561392402 1.0081143570438604 0.99
w 448 240 448 144 0
r 576 208 576 272 0 3000
w 528 208 560 208 0
w 528 272 560 272 0
d 272 128 368 128 1 0.805904783
r 176 176 112 176 0 470
r 176 304 112 304 0 470
w 208 304 256 304 0
w 192 176 256 176 0
w 112 304 112 176 0
w 112 176 112 80 0
w 448 80 112 80 0
d 192 352 368 352 1 0.805904783
w 352 160 368 160 0
w 352 320 368 320 0
w 208 128 272 128 0
w 304 240 384 240 0
z 304 240 304 304 1 0.805904783 12
r 256 176 256 240 0 10000
r 256 240 256 304 0 10000
w 304 240 256 240 0
w 176 304 208 304 0
w 384 192 384 240 0
w 192 176 192 352 0
w 176 176 192 176 0
z 304 240 304 176 1 0.805904783 12
w 256 176 304 176 0
d 384 288 384 320 1 0.805904783
d 384 192 384 160 1 0.805904783
w 368 320 384 320 0
w 368 160 384 160 0
w 352 288 384 288 0
w 352 192 384 192 0
w 224 240 256 240 0
```
I also did an alternate circuit with [Capacitor Shunt](http://carlosdelfino.eti.br/circuitjs1/?startCircuit=zvs-capacitor.txt), this worked perfectly.

The Code is below:

```
$ 1 0.000005 9.001713130052181 33 5 50
f 304 176 352 176 0 2 0.02
f 304 304 352 304 8 2 0.02
g 224 240 224 272 0
c 448 240 448 320 0 6.800000000000001e-7 -0.6705496970362219
R 512 48 512 16 0 0 40 24 0 0 0.5
l 512 80 512 144 0 0.0001 0.07090387348278972
w 512 48 512 80 0
w 448 160 528 208 0
w 448 320 528 272 0
w 448 160 384 160 0
w 448 320 384 320 0
w 384 240 384 288 0
w 368 128 368 160 0
w 208 128 208 304 0
w 256 304 304 304 0
w 368 352 368 320 0
w 448 240 512 240 0
w 512 240 512 144 0
d 272 128 368 128 1 0.805904783
r 176 176 112 176 0 470
r 176 304 112 304 0 470
w 208 304 256 304 0
w 192 176 256 176 0
w 112 304 112 176 0
w 112 176 112 80 0
w 512 80 112 80 0
d 192 352 368 352 1 0.805904783
w 352 160 368 160 0
w 352 320 368 320 0
w 208 128 272 128 0
w 304 240 384 240 0
z 304 240 304 304 1 0.805904783 12
r 256 176 256 240 0 10000
r 256 240 256 304 0 10000
w 304 240 256 240 0
w 176 304 208 304 0
w 384 192 384 240 0
w 192 176 192 352 0
w 176 176 192 176 0
z 304 240 304 176 1 0.805904783 12
w 256 176 304 176 0
d 384 288 384 320 1 0.805904783
d 384 192 384 160 1 0.805904783
w 368 320 384 320 0
w 368 160 384 160 0
w 352 288 384 288 0
w 352 192 384 192 0
w 224 240 256 240 0
l 528 208 528 272 0 4 -0.07092327774163067
c 448 160 448 240 0 6.800000000000001e-7 -23.387372912945384
o 40 64 0 4099 20 0.05 0 2 40 3
o 8 64 0 4619 320 0.2 1 2 8 3
```

pfalstad, 2018-06-16:
Ok I got it working here: http://tinyurl.com/y9yg8656

but if you reset the circuit, it doesn't start back up again.  I'll look at it some more.

pfalstad, 2018-06-16:
Ok I got it to work after reset:

http://tinyurl.com/ydc3pjrg

The MOSFETs were too perfectly matched.  I changed one slightly to break the symmetry, and now it works.

carlosdelfino, 2018-06-16:
I had really forgotten to put a resistor to complete the reactance of the coil. I do not know how to intervene in the code, but this resistor should already be intrinsic to the inductor, in addition there is the coefficient of resistive permeability of the core, in my case this core is variable, since it will be inserted for induction heating ..

I made some adjustments in the resistors and was happy:

```
$ 1 0.000005 35.60246606707791 45 5 50
f 304 176 352 176 0 2 0.02
f 304 304 352 304 8 2 0.02
g 224 240 224 272 0
c 416 160 416 320 0 6.800000000000001e-7 24.599557969458573
R 448 48 432 16 0 0 40 24 0 0 0.5
l 448 80 448 128 0 0.00019999999999999998 3.5527136788005016e-16
w 448 48 448 80 0
w 416 160 384 160 0
w 416 320 352 320 0
w 384 240 384 288 0
w 368 128 368 160 0
w 208 128 208 304 0
w 256 304 304 304 0
w 368 352 368 320 0
169 544 272 464 272 2 0.4 1 1.734723475976807e-18 -0.05735680263880661 -0.057356802638806624 0.99
w 448 240 448 176 0
w 544 208 560 208 0
w 544 272 560 272 0
d 272 128 368 128 1 0.805904783
r 176 176 112 176 0 470
r 176 304 112 304 0 470
w 208 304 256 304 0
w 192 176 256 176 0
w 112 304 112 176 0
w 112 176 112 80 0
w 448 80 112 80 0
d 192 352 368 352 1 0.805904783
w 352 160 368 160 0
w 352 320 368 320 0
w 208 128 272 128 0
w 304 240 384 240 0
z 304 240 304 304 1 0.805904783 12
r 256 176 256 240 0 10000
r 256 240 256 304 0 10000
w 304 240 256 240 0
w 176 304 208 304 0
w 384 192 384 240 0
w 192 176 192 352 0
w 176 176 192 176 0
z 304 240 304 176 1 0.805904783 12
w 256 176 304 176 0
d 384 288 384 320 1 0.805904783
d 384 192 384 160 1 0.805904783
w 368 320 384 320 0
w 368 160 384 160 0
w 352 288 384 288 0
w 352 192 384 192 0
w 224 240 256 240 0
r 464 272 416 320 0 5
r 416 160 464 208 0 5
r 448 176 448 128 0 5
o 40 64 0 4107 20 0.05 0 2 40 3
o 16 64 0 4099 80 0.00009765625 1 2 16 3
o 48 64 0 4099 1.25 0.1 2 2 48 3
```

closed 2019-03-06 by sharpie7