sharpie7/circuitjs1 #235: Feature request: DC motor / DC brushless fan

Uzer1, 2018-07-04:
Hello. I am making a simple circuit to control my PC fans voltages depending on the temperature on a thermal resistor. 
The program Circuit Simulator 1.6i does not have a thermal resistor and an electric motor / dc fan. 
These are most common parts. I can emulate the thermal resistor [NC-type] by dragging the ohm selector in resistor options. However, the closest way I found to a dc motor is a lamp or a light diode. Is there a better way to select/emulate a dc motor? Will the motor be added to the program built-in parts selection?

parq, 2018-10-04:
+1
Please add a DC motor component.
Thank you in advance.

annopnod, 2018-10-05:
add stepper motor
https://www.youtube.com/watch?v=TWMai3oirnM

pablorq, 2018-10-24:
Ok. I'll see it later.

DmitriKhanoukaev, 2020-02-20:
Hi, I'm trying to simulate circuit with DC motor and it seems not implemented. Or is there a trick how to add it?

pfalstad, 2020-12-19:
Added DC motor contributed by Francisco Campos.

closed 2020-12-19 by pfalstad