sharpie7/circuitjs1 #237: Help - Trying to build a Memory using Custom Logic

Benur21, 2018-07-06:
I have been trying to build a 16 bits memory using Custom Logic.
It should be like a D Flip-Flop but that you can select which memory (output) you want to work on.
For example, when selecting the memory 0101, the D and Clk inputs will work like a common D Flip-Flop but only for the memory number 5. Every other memory will stay the same.
Note: No /Q being used.
This is what I am trying:
```
Inputs: N8,N4,N2,N1,D,Clk
Outputs: Q0,Q1,Q2,Q3,Q4,Q5,Q6,Q7,Q8,Q9,Q10,Q11,Q12,Q13,Q14,Q15
Definition:

?????? ABCDEFGHIJKLMNOP=ABCDEFGHIJKLMNOP
0000A+ ?BCDEFGHIJKLMNOP=ABCDEFGHIJKLMNOP
0001B+ A?CDEFGHIJKLMNOP=ABCDEFGHIJKLMNOP
0010C+ AB?DEFGHIJKLMNOP=ABCDEFGHIJKLMNOP
0011D+ ABC?EFGHIJKLMNOP=ABCDEFGHIJKLMNOP
0100E+ ABCD?FGHIJKLMNOP=ABCDEFGHIJKLMNOP
0101F+ ABCDE?GHIJKLMNOP=ABCDEFGHIJKLMNOP
0110G+ ABCDEF?HIJKLMNOP=ABCDEFGHIJKLMNOP
0111H+ ABCDEFG?IJKLMNOP=ABCDEFGHIJKLMNOP
1000I+ ABCDEFGH?JKLMNOP=ABCDEFGHIJKLMNOP
1001J+ ABCDEFGHI?KLMNOP=ABCDEFGHIJKLMNOP
1010K+ ABCDEFGHIJ?LMNOP=ABCDEFGHIJKLMNOP
1011L+ ABCDEFGHIJK?MNOP=ABCDEFGHIJKLMNOP
1100M+ ABCDEFGHIJKL?NOP=ABCDEFGHIJKLMNOP
1101N+ ABCDEFGHIJKLM?OP=ABCDEFGHIJKLMNOP
1110O+ ABCDEFGHIJKLMN?P=ABCDEFGHIJKLMNOP
1111P+ ABCDEFGHIJKLMNO?=ABCDEFGHIJKLMNOP
```
The first line keeps all outputs the same, as long as all inputs are static (the outputs only change when the clock rises; it the clock doesn't change then don't do anything).
The second and following lines, when Clk rises, will copy the input D to the selected output.

It looks like this:
[image]

But when N8=N4=N2=N1=0, D=1, and I do a rising transition on Clk, the output Q0 stays OFF. It should go ON because the memory Q0 is selected (N8=N4=N2=N1=0), D is High and Clk has rised.

Can anyone help me find what's wrong?

pfalstad, 2018-07-10:
Sorry for the late response.  Great idea!  The first line of the definition that matches is the one that is used, so you need to move the ??????? line to the end, and then it works.

Benur21, 2018-07-10:
Nice, thank you.
The Centre Circuit option does this to the circuit, though:
[image]

closed 2019-03-06 by sharpie7