sharpie7/circuitjs1 #249: Falstad Analog spdt switch

CaptainYinn, 2018-08-27:
I was creating a circuit in falstad and the circuit has a clock generator controlling an analog SPDT switch under the Active building blocks tab. The problem is that the SPDT allows current to flow through both legs regardless of whether the switch is open or closed. With this pulsed dc control the switch should be flipping between open and closed rapidly. the animation of the switch is correct but the voltage and current visualization remains powered on both legs. Thanks for your time.

pfalstad, 2018-08-27:
Are you talking about a particular circuit?  Can you send me a link to the circuit?

CaptainYinn, 2018-08-27:
http://tinyurl.com/ybg7yvds 

this is the circuit in particular, it should be switching between the two mosfet gates.

CaptainYinn, 2018-08-27:
Ahhh, that works. The reason I was questioning it was that I was able to use a manual spdt switch and could click it to create the square wave, but either way it should require the grounds. Thanks for taking the time to help, also I am assuming you have had a great deal of time invested in this program based on your name, if so I would like to thank you as this is making learning electronics and messing around a lot of fun.

pfalstad, 2018-08-27:
No problem, glad you like it!

closed 2018-09-15 by sharpie7