sharpie7/circuitjs1 #251: Clapp oscillator simulation gives wrong result

afiskon, 2018-09-09:
Hello,

I tried to use CircuitJS to simulate an oscillator I soldered not a long time ago:

[image]

The output supposed to be a low-current sinusoidal wave 10 V peak-to-peak with frequency:

```
>>> import math
>>> C1 = 1/1000/1000/1000
>>> C2 = 10/1000/1000/1000
>>> C3 = 22/1000/1000/1000/1000
>>> C3 = 1/1000/1000/1000
>>> L = 10/1000/1000
>>> 1 / (2*math.pi*math.sqrt(L*(1/(1/C1+1/C2+1/C3))))
2306374.2413629955
```

... about 2.3 MHz.  However according to simulation results this circuit doesn't oscillate. For smaller frequencies the result is the same.

Tested on https://www.falstad.com/circuit/circuitjs.html
Short URL for the circuit: http://tinyurl.com/ycpluft7

As text:
```
$ 1 0.000005 0.26073472713092677 1 5 43
R 192 64 192 32 0 0 40 5 0 0 0.5
g 192 272 192 304 0
t 160 176 224 176 0 1 -4.18279529863405 -0.025203209621387848 300
r 112 64 112 176 0 15000
r 112 176 112 272 0 4700
c 160 176 160 272 0 1.0000000000000001e-7 0.0010000004656692705
r 224 192 224 272 0 270
l 224 144 224 64 0 0.01 -0.0005089470781836736
c 304 144 304 192 0 1e-9 4.157592089012663
c 304 192 304 272 0 1e-8 0.02620321008705712
c 400 208 400 272 0 1e-9 4.183370949976207
l 400 208 400 144 0 0.000009999999999999999 -0.00026229635147379004
w 224 144 224 160 0
w 224 144 304 144 0
w 304 144 400 144 0
c 400 144 464 144 0 1.0000000000000001e-7 0.0010000000000163212
O 464 144 512 144 0
w 304 192 224 192 0
w 112 272 160 272 0
w 160 272 192 272 0
w 192 272 224 272 0
w 224 272 304 272 0
w 304 272 400 272 0
w 224 64 192 64 0
w 192 64 112 64 0
o 16 1 0 4106 10 0.1 0 1
```

sharpie7, 2018-09-09:
As with any time-series base simulation the time step needs to be small compared to the period of the highest frequency of interest. Your simulation uses a 5us time step which is too long for a 2MHz oscillator. You also forgot to connect the base of the transistor to the voltage divider that biases it.

With the circuit corrected and a 10ns time step:
http://tinyurl.com/ybkb2sv4

closed 2018-09-09 by sharpie7

afiskon, 2018-09-10:
Oops. Thank you very much for the explanation. I started to use CircuitJS only recently and apparently had some issues figuring out how to use it. Sorry for bothering!