sharpie7/circuitjs1 #264: CTRL-D in full-screen duplicates sliders

gnbl, 2018-10-08:
I tried to bookmark the fullscreen version at http://www.falstad.com/circuit/circuitjs.html by pressing CTRL-D (Firefox on Windows) but that duplicated the sliders on the right-hand control bar, although no circuit elements are selected (default circuit). Same behavior at http://lushprojects.com/circuitjs/circuitjs.html
The previous sliders then don't have any effect.

(I know the app catches keyboard input for its command shortcuts, i.e. the Duplicate command)

sharpie7, 2018-10-08:
Yes, there are various problems with cut-copy-paste-undo and duplicate with sliders. Maybe paul knows an easy fix, or I can take a look shortly. I think it all relates to probles in the dump-undump processes when one element refers to another by its position in the list of elements array. The floating scopes suffered from the same problem and I had to put in work-arounds - maybe a general solution to this problem is needed.

pfalstad, 2018-10-08:
It seemed like the sliders were safe from the bug you're describing, since you can't cut/copy/paste sliders.  That's a tricky problem you're describing, but I had some vague ideas on how to fix it.

This is a much worse problem...  Apparently all the sliders are duplicated whenever you paste or duplicate any element.  I think the fix will be much simpler though.

sharpie7, 2018-10-09:
@pfalstad Yes, the problem I was thinking of was the duplication of slideers you described, I assumed the cause was related to the point I described, but it may be due to something completely different!

pfalstad, 2019-03-06:
fixed in 993bcb68a44059d9b2f1340c54a999e98cdb167f

closed 2019-03-06 by sharpie7