sharpie7/circuitjs1 #266: Voltage from nowhere!

sharpie7, 2018-10-15:
Flipping the switch a few times in this circuit will cause strange voltages to appear in the LED branch when the switch is left on the lamp side.

[http://tinyurl.com/y7mrz2sj](http://tinyurl.com/y7mrz2sj)

It appears to require both the lamp and the MOSFET to be present to create this effect.

Bort-Plate, 2018-10-23:
I see the LED branch flashing red a couple times, but it doesn't show on a scope graph. Perhaps just a graphical error?

pfalstad, 2018-10-23:
Ok that's weird.  I'm seeing it on the scope graph.

Bort-Plate, 2018-10-23:
Correction: It *does* show up on the scope for that horizontal piece of wire at the end, as well as both diodes. Not for the resistor though. It actually appears regardless of any interaction, and the same waveform shows at the same time signature after each reset. First occurrence on the top LED seems to be at about 15ms.

pfalstad, 2018-10-23:
You can replace the lamp with a varying resistor.  Start the resistor out at 1 ohm while the LED's are connected, then switch over, then increase the resistance to larger values and the LED's will go nuts.  If an LED is dragged somewhere else when it's in this state, it still has phantom voltages across it.

pfalstad, 2018-10-23:
My guess is it's a badly conditioned matrix.  The LED's have a larger voltage drop which causes very small values to be loaded into the matrix, as small as 1x10^-24, and that can cause numerical issues, but it still seems like it shouldn't be this bad.

E3V3A, 2019-02-26:
If you use the normal LED settings, it works. For some reason the LEDs you have set to `fwdrop=3` whatever that means. Also set the MOSFET Beta to ~`60` and not `50m`.

pfalstad, 2019-02-26:
This has been fixed, assuming you use the normal LED settings.

E3V3A, 2019-02-26:
Ah, ok, so what are the not normal settings available and where are they described?
Hard to help if the OP doesn't say he's using non standard settings on purpose. (Double negations there!) :tada:

pfalstad, 2019-02-26:
This is an old bug from before it was fixed.  Don’t need to help OP, he can take care of himself. 🙂

E3V3A, 2019-02-26:
Sure I'll get out of your way. But since this is a public repo it's impossible to know in advance who knows what and who doesn't. In any case, since the model linked above is still showing an issue, I'm curious to know what you mean when you say it has been fixed? And if it has, why is this issue still open?

pfalstad, 2020-12-11:
Sorry I didn't respond to you a year and a half ago.  It's fixed if you use the corrected LED model.  We can't reasonably fix it to work with the bad LED model because we only have so much numerical precision.

closed 2020-12-11 by pfalstad