sharpie7/circuitjs1 #267: Confusing FET and BJT operation region names

aspenforest, 2018-10-16:
Greetings.

The current descriptions of FET and BJT operation regions are confusing, because 
when one tries to design a circuit in the simulator using tutorial schematics from the internet,
multiple sources have named the FET and BJT regions differently. 

I propose that the cutoff state be named "off state", the resistive state be named "ohmic state", and 
the fully turned on state be renamed "active state".

Felthry, 2018-10-30:
There are very well-entrenched names for these regions in the literature; for BJTs, you have cutoff, forward active, reverse active, and saturation; and for FETs you have cutoff, linear, and saturation. I don't think you're going to get people to change those anytime soon.

sharpie7, 2019-03-06:
Agreed with @Felthry the names, for better for worse, are the industry standard names so we will retain them.

closed 2019-03-06 by sharpie7