sharpie7/circuitjs1 #269: Feature: non-superconducting transmission lines.

ormaaj, 2018-10-27:
The light-speed transmission lines aren't very useful for most applications. It would be convenient to have fields for specifying a velocity factor and length, and have the delay automatically calculated. Perhaps with a default VoP of .65-.66 for typical RG-58 cable (looks like the current default impedance is 75 ohm).

For extra credit, add attenuation / losses. Nonlinear frequency response complicates that though, and it might be difficult to come up with a reasonably generic model.

ormaaj, 2018-12-04:
a0f63e4dd5a1c3e74c3491b185868110a5b1b3ec ...Ah so it was just the displayed value.

pfalstad, 2018-12-06:
Hi, I just fixed the displayed value for now, but later we can change it so you can create transmission lines with a specified length (instead of specifying delay).