sharpie7/circuitjs1 #280: Bug - Current to nowhere!

Chisight, 2018-12-05:
This nonsensical circuit http://tinyurl.com/ya847a6e is the minimal example that I was able to create showing the problem.  It was pared down from a zener protected high side mosfet switch that exhibited odd characteristics at higher voltages but not lower ones.
To demonstrate the issue, tap the momentary switch and note that once released the current going into the bipolar and resistor is two orders of magnitude higher the current going out!  
Reset by opening and closing the lower switch and currents return to normal until the momentary switch is pressed again.
Lower the 60V source to 55V and the problem goes away.
Removing the mosfet's ground and moving the 60V and 1K input to take the zener and mosfet out of the active circuit will still demonstrate the problem to a lesser degree but removing the inactive components removes the problem. (http://tinyurl.com/ychl56m8 is the alternate circuit, note no current flowing to/from the mosfet and zener.)

E3V3A, 2019-02-26:
I can't see the problem. Can you please describe what is expected vs what is happening?
In fact your cicuit is somewhat similar to my own in a different issue, that now works.

However, at any voltage >~12, you have reached the zener breakdown at ~5V because of the Vr drop of 6.3 V. So you then its a question how well that region is simulated. I suggest you edit your simulation and add a few scopes to it and describe better what you think is wrong. It could be that the time-step need to be tuned. I noticed that your original circuit gave a huge CPU load, which indicate that something is going out of control in the simulation equations. I reduced the timestep by /10 and reduced the simulation speed, and don't see any immediate problem.

I don't see this current you talk about, please show us on the scope!
(Also, don't get confused by the movement of the little "current" dot's, they can appear going in reverse in some conditions or not move at all, even when they should.)

nietamten, 2019-03-29:
In second example ( http://tinyurl.com/ychl56m8 ) if I delete zener it works much different and it looks wrong.
I found something maybe similar http://tinyurl.com/y2vot74g .

trlolo, 2019-04-17:
[circuit-20190417-1458.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/3089473/circuit-20190417-1458.circuitjs.txt)
no coments

E3V3A, 2019-04-19:
Well guys!
**`No screenshot + No explanation = NO HELP!`**

pfalstad, 2019-04-22:
I see the issue..  I will try to figure out what's going on.

pfalstad, 2019-04-22:
There were a couple of problems here..  The transistor was calculating the incorrect current (and the nearby wires were using that bad current).  Also, the bad current was making it harder for the simulation to converge.  So, thanks for reporting this.

pfalstad, 2019-04-22:
trlolo's example isn't related to this..  I replaced the 1 picoohm resistors with 1 ohm and it stopped acting weird.

Chisight, 2019-06-18:
Sorry for the delay in responding, the status emails were delivered to Spam.
I can confirm that the problem no longer exists in the current hosted version.  

Thank you for the fix!

closed 2019-06-18 by Chisight