sharpie7/circuitjs1 #282: Current Source & Ampmeter Direction when NOT using Convetional Current Motion

AncientMist, 2018-12-05:
Hi, 

I turn-off the Conventional Current Motion since I'm teaching electron flow theory. When I do this, the ammeter and the current source shows the current flow (pointed arrow) in a different direction. Can this be automatically flipped when this Convention Current Motion is unselected?

For the ammeter, one solution is to change the symbol. Maybe use an "A" symbol with a plus and minus symbol on both ends. This would eliminate the need to flip the visual diagram.

pfalstad, 2018-12-06:
ok will fix.

sharpie7, 2018-12-06:
I have a different opinion on this. The convention in conventional current flow is that current sources and meters are labelled with the conventional current, not the electron current. While it is fine to show the electron current in the motion I think labelling components with electron current breaks the point of the convention and is more confusing than the current situation.

pfalstad, 2018-12-06:
That is a good point.  But what about changing the symbols do they don't include an arrow?

sharpie7, 2018-12-06:
For the meter I have no problem with that suggestion. For the current source then the arrow is part of the standard symbol so I would prefer to keep it, though we could aditionally label + and - which would make the fact that the arrow show conventional current clear.

pfalstad, 2018-12-06:
Well I was thinking of one of these symbols, like c which is easy to draw, or d which I've seen in circuits before..  https://electronics.stackexchange.com/questions/373569/where-does-this-current-source-symbol-come-from

But I would keep the symbols the same as they are now unless conventional current is unchecked.

AncientMist, 2018-12-17:
Hi. Just following up. 

Can the amp meter be changed to the symbol +(A)- as mentioned? This would cover both conventional and electron flow as the same symbol, which is also a standard symbol notation.

For the current source, I see where this would be a little confusing. I'm fine with leaving the symbol as is but can you include a + & - to note polarity direction and then flip the arrow when Convential Current is unchecked?

See the attached for visuals on the mentioned change request.
[image]

pfalstad, 2018-12-17:
I was proposing to change the ammeter symbol to +(A)- when conventional current is unchecked, and leaving it the same when conventional current is checked.

For the current source, we were saying we don't want to flip the arrow.  We were going to change it to symbol "c" here: https://electronics.stackexchange.com/questions/373569/where-does-this-current-source-symbol-come-from  but only when conventional current is unchecked.

AncientMist, 2018-12-19:
Sounds goods. If you can add the + & - for polarity verification then this would make it clear. See the reference picture.

[image]

pfalstad, 2019-03-07:
Doesn't the (conventional) current flow from - to + ?  Your attached visuals are showing it flowing the opposite direction.

AncientMist, 2019-03-21:
(Conventional) current flows from + to - . (Electron) current flows from - to + . 

The snapshot pic would have the polarity symbols on the current source to know where it gets hooked up in the circuit.

pfalstad, 2019-03-21:
well, I created an A/C source with a large DC offset and a small max voltage in the simulator, and the positive (voltage) side of the voltage source is marked +.  Conventional current flows from the - side to the + side and out into the circuit.

Similarly, for a current source, conventional current flows out of the side that has the higher voltage (which I assume should be marked +).