sharpie7/circuitjs1 #287: Bug: Custom Logic Not Loading From Save

SgtSchultz71, 2018-12-17:
I can build a variety of custom logic circuits without any issue. The problem occurs when I attempt an auto-recover or loading a save from a file. My custom logic chips will disappear from their original spots! The weird thing is that I can drop a new custom logic down, rename it as an original one, and the original custom logic data will return as if it WAS saved.

SgtSchultz71, 2018-12-18:
I just did an experiment and found that if I rename my custom logic chips to one word names, they save just fine. All the chips that didn't reload had two-word names. I can still send you the file if you want, but I thought that was interesting.

pfalstad, 2018-12-18:
Ok, I see the problem now.

pfalstad, 2020-09-10:
Fixed on 12/18/18

closed 2020-09-10 by pfalstad