sharpie7/circuitjs1 #289: Colors

SvenNovgorod, 2019-01-09:
The solar batteries I have use Red for the positive terminal and Blue for the negative. My car has red for the positive terminal of it's battery. All the DC equipment I have repaired use Red for the positive terminal and black for negative. I did find it confusing you have used Red to show negative voltages on the animation. You may of course have a good reason for this but I have spent an hour trying to work out what was happening assuming green was negative and red positive.

Apart from this issue it's a Great Program

B-URu, 2019-06-14:
Dear Paul Falstad,
this program (and also all the other applets) is absolutely great! I love to use it for my students to simulate simple and even more advanced circuits in the context of a physics lecture. It would be very helpful to be able to change the color code, because it conflicts with the usual color code used for electric charges and magnetic poles. In my observation the most common color code is (red=positive, blue=negative) for electric charges and (red=north pole, green=south pole) for magnetic poles. The color codes available in the applet are somehow a mixture of these most common codes. Would it be versy difficult to change that?
Best regards,
Bernd-Uwe Runge

pfalstad, 2020-12-19:
Colors can be changed now in Options->Other Options or the URL.

closed 2020-12-19 by pfalstad