sharpie7/circuitjs1 #290: [Feature Request] allow to make status information stick

herrmanthegerman, 2019-01-15:
Hello,

I have this small example circuit:
[image]

If you move the mouse over any component of the circuit, you will see a short description of that component and its current state in the lower right corner of the window. In the screenshot the mouse hovers over the highlighted transistor and consequently its state information is displayed.

However, if I move the mouse away to control the slider in order to change the resistor value, the component state changes to the information of the resistor.

That's fine, however, in addition to the resistor's state information I'd like to see the transistor's state information in parallel. So I can see, e.g., how the resistor value affects power dissipation of the transistor.

Maybe this is already possible and I just do not know how to do it. Otherwise, I'd like to submit this as a feature request.

And thanks a lot for this superb tool!

E3V3A, 2019-02-26:
I agree. That would have been a useful feature to allow it to be static, via some checkbox somewhere,

pfalstad, 2019-02-26:
You could just plot it in a scope.  Well, right now you can't display a transistor's power in a scope, but I'll fix that.

If you want to see the actual value instead of a graphical representation, I could add it to the scope easily.  Maybe the whole info display could be added to the scope.

We already have a UI for adding and removing scopes, embedding them in a circuit, etc. so that seems to be the simplest solution.

pfalstad, 2022-01-15:
now you can put that info in the scopes.  hide the graphs if it's too hard to read.  https://tinyurl.com/yahewaf4

closed 2022-02-19 by pfalstad