sharpie7/circuitjs1 #293: identification of the component and add datasheets specifications

miguelfernando55, 2019-01-23:
Hi,
I would like to suggest that all components could be identify to as an ideal component, but it behavior chould be updated to have datasheet specifications, respective to a real datasheet.

E3V3A, 2019-02-27:
I think this is a good idea, at least so for the following types of components, that could have drastically different behaviors depending on their details. It would make it much easier for inexperienced users to troubleshoot issues.

- Active Components
- Active Building Blocks
- Analog and Hybrid Chips

caracole, 2019-07-09:
I suggest the default identification of the components ie a 'name' in the properties window followed by an automatic index if the component is duplicated. Ex: for a relay K1, K2 etc
Similarly, assign a connection number eg for a relay: K1 (relay name) A1, A2 (the coil) 11,12,14; 21,22,24 Contacts etc
I do not think these changes are complex, but they will help to enrich the software.
(French to English translation by google translate)