sharpie7/circuitjs1 #298: amateur radio

johnmatsson, 2019-02-12:
I would love to see other inputs except AM and FM, primarily SSB. One can then try different product detectors, demodulators. 

Also another component, a dual gate mosfet, i know its possible to use two single gate mosfets instead, this also means that the work behind making a dual gate mosfet shouldn´t be that big. 

Thanks for a fantastic simulator, i love it.

E3V3A, 2019-02-25:
@johnmatsson 
These are two different issues. Please file separate issues with as many relevant details as possible, otherwise it is very unlikely anyone will be able to implement them.

pfalstad, 2019-02-26:
You can use a VCVS to generate any signal you want, if it can be expressed as a function.