sharpie7/circuitjs1 #301: Broken MOSFET behaviour

E3V3A, 2019-02-25:
I love this highly useful tool, BUT...

After having spent considerable (too much) time trying to get a simple *"High-Drive"* circuit to work, I basically kept on failing until I found the **`Threshold Voltage`** and **`Beta`** settings, and set them to **Zero**. This probably doesn't make any sense, and the Voltages produced seem very inconsistent. 

Here is the [circuit](http://tinyurl.com/yyj238wx):

[image]

And here is the code:
```
$ 1 0.000005 114.32915054483153 38 5 50
R 176 240 144 240 4 5 2 5 0 0 0.5
O 400 16 448 16 0
r 240 16 240 96 0 4700
R 176 16 128 16 0 0 1 12 0 0 0.5
f 320 64 320 16 9 -1 1
w 240 16 304 16 0
w 320 64 320 96 0
w 320 96 240 96 0
w 320 96 320 128 0
t 272 240 320 240 0 1 -11.999849610521181 5.263992931596369e-11 100
w 320 256 320 288 0
r 208 240 272 240 0 470
w 176 240 208 240 0
g 320 288 320 304 0
w 320 224 320 176 0
w 336 16 400 16 0
w 176 16 240 16 0
162 320 128 320 176 2 default-led 1 0 0 0.01
x 348 150 379 153 4 12 LED1
x 346 165 395 168 4 10 HVSP_ON
x 355 47 369 50 4 12 T1
x 333 240 347 243 4 12 T2
x 204 60 220 63 4 12 R1
x 234 265 250 268 4 12 R2
x 60 254 109 257 4 10 HVSP_ON
x 62 238 83 241 4 12 \p5V
x 398 8 433 11 6 10 RESET
x 150 2 170 5 4 10 BAT
368 176 240 176 176 0 0
g 400 96 400 112 0
w 400 16 400 48 0
r 400 48 400 96 0 1
o 28 1024 0 4610 5 0.1 0 2 28 3
o 1 1024 0 4610 20 0.1 0 1
38 4 0 1 1000 Threshold\sVoltage
38 4 1 1 1000 Beta

```

aheid, 2019-02-25:
Since it's a p-channel MOSFET, the threshold voltage needs to be negative. Setting it to -1.5 (realistic for logic level p-MOSFET) or even -4 (power p-MOSFET) and the circuit works as expected.

Of course, the simulator shouldn't accept garbage input such as positive threshold voltages for p-MOSFETs.

E3V3A, 2019-02-25:
@aheid 
Fantastic! Thank you, that finally solved my (and other peoples?) problem!

So basically we have the following problems with the **`p-MOSFET`** item:

1. The *slider* **`Beta`** for a **p-MOSFET** should be in the range:  **`[1,80]`**
2. The *default* should be **`~60`** ?
3. The *slider* **`Vthreshold`** for p-MOSFET should be in the range:  **`[-4,1]`**
4. The *default* should be **`-1.5`**

Similarly other default values should be implemented for the n-MOSFET?

pfalstad, 2019-02-26:
There’s no sliders, unless you’re talking about the java version, which I’m not updating anymore.

The p-mosfet threshold default is already -1.5.  We had a discussion about the default beta a while back, and finally decided to leave it as is, even though it’s way too low for power mosfets.

E3V3A, 2019-02-26:
@pfalstad 
> unless you’re talking about the java version, which I’m not updating anymore.

Huh!? I'm only aware of the ones on you guys have on your websites. That's what I use and presumably 99% of all others using this tool. It's a super great convenience to be able to link to your sites and run something instantly. It would be a great loss and sadness to see those disappear.

pfalstad, 2019-02-26:
I mean the java version at http://www.falstad.com/circuit-java/ ... I'm not updating that.  It has sliders in the edit dialogs, but the current one at http://www.falstad.com/circuit/ does not.

E3V3A, 2019-02-26:
@pfalstad 
>  but the current one at ... does not.

Sure it does. Select a part, e.g. p-MOSFET, right-click and select `sliders..`.

pfalstad, 2019-02-26:
Oh right.  Well, in that case the defaults are always 1 to 1000.  You can change them to an appropriate range yourself.

E3V3A, 2019-02-26:
Paul, that is exactly what I'm saying. It took me 2 days and only thanks to a random comment by Asbjørn, was I able to resolve this. The default settings should be something that make sense, and not something that doesn't work. Default range of 1-1000 when you need a negative number? You had even forgot about it yourself. 

This should be easy as a few clicks to implement. Just put in the values above and most people will probably be happy and many other masked MOSFET issues may get resolved. 

Also, again refering to what I said in other issue. If user can't see or find a Beta in the datasheet, he will most likely be clueless as what to put in the Beta field, while in addition fooled by the default values (as I was). That is also true for Vth.

pfalstad, 2019-03-07:
Actually positive thresholds for a p mosfet (or negative for n channel) are just fine.  Those are depletion mode mosfets.  There are some in the ternary logic examples.

pfalstad, 2019-03-07:
I’m adding a link to a worksheet to calculate beta from the mosfet edit dialog.

pfalstad, 2020-12-11:
So it sounds like the only probem here is the default values for the sliders being bad, which is filed as #312.

closed 2020-12-11 by pfalstad