sharpie7/circuitjs1 #304: Improvement of in circuit part descriptions

E3V3A, 2019-02-26:
This is a suggestion to improve the UX, in particular beginners, for finding more information about how certain parts works. It seem that this idea is already in there. For example, clicking EDIT on a VCVS part, we find a link in top of the window to: http://lushprojects.com/circuitjs/customfunction.html, however, this link leads nowhere. It would be great if this could be linked to either a local web server, Wikipedia or other maintained place. In addition, we may also consider making a simple pop-up text-box, very briefly describing how the part works and what it is used for.

sharpie7, 2019-02-26:
We already have a managed source for the in-application documentation. It is this repository. Updates to the in-application documentation can be made by PR or Issues here.

here is the file:
https://github.com/sharpie7/circuitjs1/blob/master/war/customfunction.html

I will fix the broken link on my server. Thanks for that.

closed 2019-02-26 by sharpie7

E3V3A, 2019-02-26:
I think the feature was more general than just that one part. 
It would be great to have a page (or even a JSON driven info-box) for every part.