sharpie7/circuitjs1 #305: Add github Wiki entry about how to add new parts?

E3V3A, 2019-02-26:
Perhaps this is too late to the party, but seeing that many people request new parts to be added , perhaps it would be more helpful to add some additional (markup) info somewhere (perhaps as a github WIki) on how to add new parts to your part-list. That way the dev's can get new parts with less dev effort. 

What are your thoughts?

sharpie7, 2019-02-26:
There are a lot of aspects to adding new parts and actually coding them isn't the hard bit. Designing a model that will be a reasonably good representation of what people expect from real parts while also behaving correctly in the simulation can be very complex. This is particularly true for some of the parts on the request list which don't have industry-standard models associated with them.

How to actually add a new part to the program is reasonably obvious by looking at the code once you have understood how to model it. Quite a few people have already contributed new parts to the project.

To be blunt - I think that people who have the skills to be able to develop new parts won't have much trouble understanding how to add them to the code.

E3V3A, 2019-02-26:
> How to actually add a new part to the program is reasonably obvious by looking at the code once you have understood how to model it.
>  I think that people who have the skills to be able to develop new parts won't have much trouble understanding how to add them to the code.

Ok, sound good. I must admit the code is still a mystery to me, so feel free to close the issue.

closed 2019-02-27 by sharpie7