sharpie7/circuitjs1 #306: MOSFET (and others) doesn't show pin names when not in horizontal position

E3V3A, 2019-02-26:
If you turn (in this example) the MOSFET so that the *gate* pins are not horizontally located, then all the pin names disappear. This is unfortunate and seem buggy.

[image]
[image]

ormaaj, 2019-03-09:
OTOH the busy MOSFET graphic is hard to distinguish when there are lots bunched together. Labels add clutter and the orientation is unambiguous without them. If only the body diode feature still worked when "show bulk" is unchecked. The digital symbol with the body terminal not connected to the drain is really common in IC schematics, and sometimes easier to read.

pfalstad, 2019-03-09:
This is just laziness on my part..  I've fixed it in my sandbox.

The gate pins are only labeled when the MOSFET is selected, so I didn't think it added that much clutter?

We could have the body diode work when "show bulk" is unchecked, but where would the body be connected?  I thought in IC's, the body terminals are mostly connected in common to the most negative or most positive potential, so we don't have to worry about body diode current.  That's basically the same as not simulating the body diode.

ormaaj, 2019-03-09:
> The gate pins are only labeled when the MOSFET is selected, so I didn't think it added that much clutter?

Oh true. Not really an issue I guess.

> We could have the body diode work when "show bulk" is unchecked, but where would the body be connected? I thought in IC's, the body terminals are mostly connected in common to the most negative or [...]

I'm really just learning this but have seen a few uses that look more involved. E.g. CMOS bandgap reference topologies that play tricks with the body voltage to attain a threshold that's +/- some offset.

There's this SPDT analog switch (Page 3): https://www.vishay.com/docs/70051/dg417.pdf that toggles the body of the N-FET on the right between some inputs that can be set independently of all the others. Not quite the same as the basic transmission gate. I guess there are multi-supply scenarios where it could be used to some effect. Or I'm wrong and it isn't so interesting - have to play with it more... :confused:

pfalstad, 2019-03-09:
Interesting..  Well, at least for the SPDT switch, you definitely need a body terminal to see the fancy things they're doing with the body diode.  Using a digital MOSFET symbol wouldn't be appropriate (which is why page 3 doesn't use them).

E3V3A, 2019-03-09:
Although your discussions are very important and interesting, it seem totally OT for this one. 
Please open a new issue for what you are discussing. 
Here, the only thing I ask, is that the pin names show when the mouse is help over the component, so nothing fancy at all.

E3V3A, 2019-03-13:
@pfalstad 
I found all your new fixes! Fantastic. 
:+1:

closed 2019-03-13 by E3V3A